earlephilhower/ESP8266Audio #199: Play WAV from SD

NickVerlooy93, 2019-07-23:
I'm trying to play a WAV file from the SD card. I'm using the ESP LyraT V4.3 board. 

`#include <Arduino.h>
#include <SPI.h>
#include "mySD.h"
#include "AudioFileSourceSD.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2SNoDAC.h"

AudioGeneratorWAV *wav;
AudioFileSourceSD *file;
AudioOutputI2SNoDAC *out;

void setup()
{
  Serial.begin(115200);
  delay(1000);

  if (!SD.begin(13,15,2,14))
  {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
  delay(1000);

  file = new AudioFileSourceSD("beepWAV.wav");
  out = new AudioOutputI2SNoDAC();
  wav = new AudioGeneratorWAV();
  wav->begin(file, out);
}

void loop()
{
  if (wav->isRunning()) {
    if (!wav->loop()) wav->stop();
  } else {
    Serial.printf("WAV done\n");
    delay(1000);
  }
}`

I'm getting the following error message: 

Arduino: 1.8.9 (Windows 10), Board:"ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"
In file included from C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:21:0:

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.h:44:5: error: 'File' does not name a type

     File f;

     ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp: In member function 'virtual bool AudioFileSourceSD::open(const char*)':

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:34:3: error: 'f' was not declared in this scope

   f = SD.open(filename, FILE_READ);

   ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:34:7: error: 'SD' was not declared in this scope

   f = SD.open(filename, FILE_READ);

       ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:34:25: error: 'FILE_READ' was not declared in this scope

   f = SD.open(filename, FILE_READ);

                         ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp: In destructor 'virtual AudioFileSourceSD::~AudioFileSourceSD()':

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:40:7: error: 'f' was not declared in this scope

   if (f) f.close();

       ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp: In member function 'virtual uint32_t AudioFileSourceSD::read(void*, uint32_t)':

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:45:10: error: 'f' was not declared in this scope

   return f.read(reinterpret_cast<uint8_t*>(data), len);

          ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp: In member function 'virtual bool AudioFileSourceSD::seek(int32_t, int)':

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:50:8: error: 'f' was not declared in this scope

   if (!f) return false;

        ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:51:29: error: 'f' was not declared in this scope

   if (dir==SEEK_SET) return f.seek(pos);

                             ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:52:34: error: 'f' was not declared in this scope

   else if (dir==SEEK_CUR) return f.seek(f.position() + pos);

                                  ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:53:34: error: 'f' was not declared in this scope

   else if (dir==SEEK_END) return f.seek(f.size() + pos);

                                  ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp: In member function 'virtual bool AudioFileSourceSD::close()':

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:59:3: error: 'f' was not declared in this scope

   f.close();

   ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp: In member function 'virtual bool AudioFileSourceSD::isOpen()':

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:65:10: error: 'f' was not declared in this scope

   return f?true:false;

          ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp: In member function 'virtual uint32_t AudioFileSourceSD::getSize()':

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:70:8: error: 'f' was not declared in this scope

   if (!f) return 0;

        ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:71:10: error: 'f' was not declared in this scope

   return f.size();

          ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp: In member function 'virtual uint32_t AudioFileSourceSD::getPos()':

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:76:8: error: 'f' was not declared in this scope

   if (!f) return 0;

        ^

C:\Users\nick.verlooy\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:77:10: error: 'f' was not declared in this scope

   return f.position();

          ^

Meerdere bibliotheken gevonden voor "WiFiClient.h"
Gebruikt: C:\Users\nick.verlooy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi
Niet gebruikt: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Fout bij het compileren voor board ESP32 Dev Module

Dit rapport zou meer informatie bevatten met
"Uitgebreide uitvoer weergeven tijden compilatie"
optie aan in Bestand -> Voorkeuren.

Some suggestions how to solve this problem?

Avon1, 2020-06-05:
Send a buffer to this
Wrong

- `wav->begin(file, out);`

Correct

- `wav->begin(id3,out);`

- `id3 = new AudioFileSourceID3(file);`

need to pass buffer not file I think
look at the mp3 example

Avon1, 2020-06-06:
It works with esp32 you made the above mistake see my code below
`#include <Arduino.h>`
`#include "AudioFileSourceID3.h"`
`#include "AudioOutputI2SNoDAC.h"`
`#include "AudioGeneratorWAV.h"`
`#include "AudioFileSourceSD.h"`
`#include "FS.h"`
`#include "SD.h"`
`#include "SPI.h"`

`AudioFileSourceSD *source = NULL;`
`AudioGeneratorWAV *wav = NULL;`
`AudioOutputI2SNoDAC *out = NULL;`
`AudioFileSourceID3 *id3 = NULL;`

`void setup() {`
`  Serial.begin(115200);`
 ` Serial.println();`
`  if (!SD.begin()) {`
 `   Serial.println("Card Mount Failed");`
 `   return;`
  `} else {`
    `Serial.println("SD card is ready!");`
  `  uint64_t cardSize = SD.cardSize() / (1024 * 1024);`
`    Serial.printf("SD Card Size: %lluMB\n", cardSize);`
`  }`
`  //read only wav files in SD root directory`
`  File root = SD.open("/");`
 ` if (!root) {`
  `  Serial.println("Failed to open directory");`
   ` return;`
  `}`
  `if (!root.isDirectory()) {`
   ` Serial.println("Not a directory");`
   ` return;`
 ` }`
 ` File file = root.openNextFile();`
`  while (file) {`
`    if (String(file.name()).endsWith(".wav")) {`
     ` Serial.print("FILE: ");`
      `Serial.print(file.name());`
  `    Serial.print("  SIZE: ");`
 `     Serial.print(file.size()/(1024*1024));`
  `    Serial.println(" MB");`
  `    playwav(file.name());`
 `   }`
  `  file.close();`
 `   file = root.openNextFile();`
 ` }`

}
`void playwav(String path) {`
 ` source = new AudioFileSourceSD();`
`  source->open(path.c_str());`
 ` id3 = new AudioFileSourceID3(source);`
`  //  id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");`
 ` out = new AudioOutputI2SNoDAC();`
`  wav = new AudioGeneratorWAV();`
`  wav->begin(id3, out);`
`  while (wav->isRunning()) {`
`    if (!wav->loop()) wav->stop();`
`  }`
`  Serial.println("End of playback");`
`  source->close();`
`}`
`void loop() {`

`}`

earlephilhower, 2020-06-06:
Thanks, @Avon1 .  He also missed some required includes which your example has, but yes you can make a buffer in front of the SD card like you described.

closed 2020-06-06 by earlephilhower