earlephilhower/ESP8266Audio #200: Soundfont library update

AlexFundorin, 2019-07-25:
The original one is v0.8 now. If there are some important changes, update, please.
https://github.com/schellingb/TinySoundFont/blob/master/tsf.h

earlephilhower, 2019-07-25:
The TSF lib porting is done, so you can try your luck writing the code to work with a MIDI controller.  The current setup uses a .MID file player and precomputes things, so I don't think it's straightforward to make it work with real-time inputs.

TSF.h itself has only minor updates from what I see on the commit log.  If you'd like to make PR, I'd be happy to look at it.

AlexFundorin, 2019-07-25:
Thanks. If I would be able to make the simple example work, I'd definitely try it with a real-time MIDI input.
I'm currently using a dedicated synth board to output sound from Arduino with RT-MIDI but willing to try using ESP32's own capabilities instead.

closed 2019-07-25 by AlexFundorin

campidelli, 2024-09-04:
@AlexFundorin and @earlephilhower  I made something like that, but the audio lags a little. You can check it out here
https://github.com/campidelli/arduino-accordion/tree/tiny-sound-font

Unfortunately, I don't have the skills to improve it, my knowledge of C++ and microcontrollers is quite limited.

Cheers!