earlephilhower/ESP8266Audio #201: Easy way to test the library

AlexFundorin, 2019-07-25:
I was hoping to quickly try if I can run midi file and SF2 from SPIFFS through headphones, connected to onboard ESP32 DAC (GPIO25 and 26).
I took the example (furelise), changed AudioOutputI2S.h include to AudioOutput.h (also did that to 'dac' object), cause I don't need anything fancy, just to check if it's working at all.
Then, upon building the sketch in platformio with VSCode, I got "SD and HttpClient libs not installed" errors.
NP, I reinstalled those libs. Actually, tried installing different ones in different order. Still the same errors.
I ended up manually downloading them and putting into 'lib' folder, adding the corresponding lines to platformio.ini (lib_deps).
Then, upon building, I got tons of errors, like 

```
lib\ESP8266Audio\src\AudioFileSourceHTTPStream.h:55:5: error: 'HTTPClient' does not name a type
lib\ESP8266Audio\src\AudioFileSourceHTTPStream.cpp:139:10: error: 'http' was not declared in this scope
lib\ESP8266Audio\src\AudioFileSourceICYStream.cpp:41:3: error: 'http' was not declared in this scope
```

The thing is that I'm not using neither HTTP nor ICY or SD in my test sketch and I don't get why they are even mentioned. Even if I would use some HTTP stuff, it would be with AsyncWebServer library and not with HttpClient. And I won't use the web server for playing audio/midi.
All I want is to simply play a midi file with some small soundfont, as a proof of concept and, if it would work, buy an external DAC and SD card adapter for my project.
Why is this so complicated?

earlephilhower, 2019-07-25:
Can you build an unmodified example program?  The PlayWAVFromPROGMEM example is self contained.

If you can't build it, then there is likely some issue with the Arduino installation.  The classes you metioned are base classes provided by the Arduino core libs.  Have you tried the plain Java Arduino IDE?  Might be VSCode doesn't support normal builds.

AlexFundorin, 2019-07-25:
Output: https://i.imgur.com/eb0BxZI.png
HttpClient is installed: https://i.imgur.com/FMpiQn1.png

As for Arduino IDE, I don't use it, yet, tried now. It was complaining that I have multiple WiFi and SD libraries installed.
I found one Arduino SD library installed, which, I suppose, should be used with ESP8266Audio lib and another one by Teensy board developer, though it isn't called SD, but SerialFlash. It might cause issues, but I'm actually using that library in my other project and I don't want to delete it, even though I'm not using Arduino IDE anymore. Just in case.
https://i.imgur.com/2XEeFlu.png

As for wifi lib, I only have this ones installed and I need all of them: https://i.imgur.com/5csr83V.png

The thing is that I don't have any issues with any other libs in platformio, except, maybe, one, for which I found a workaround in it's repository's issues section. It had issues with Time library.
Every other library is working fine and I'm using >10 of them in a single project.

closed 2020-08-08 by earlephilhower

PoulpePalpitant, 2021-07-08:
I had the same issue, and for some reason, including the HTTPCLient.h Arduino library manually solved the problem somehow.
[image]

I simply added this line to main.cpp:
#include <HTTPClient.h>