earlephilhower/ESP8266Audio #202: Audio synchronisation between multiple devices

daniel2130, 2019-07-31:
Hi, I am newly using your library and i have to say its a great job you did. 
I want to make some devices like google home to only stream music in the rooms actually i wanna build some speakers based on esp8266 audio using esp32 but i am looking on a solution on how i could synchronise the playback of the stream throught all the devices i am currently using mqtt to broadcast on my network the metadata and control the volume of the playback i was thinking about using a local ntp server to keep every devices on sync to play music together like google home you can specify that one speaker have 92 ms of delay to delete echo effect between rooms do you have any idea how i could begin with it i currently disabled the buffer and it bring the delay about half a second so i have to find a solution to delay or skip the playback in the case that the devices connect with some lag or delay between them. 

Thanks a lot !

percramer, 2020-06-21:
If anybody finds a solution for this, then please share! I am interested in synchronisation also

earlephilhower, 2020-08-10:
This is outside of the audio domain and into high accuracy (millisecond or less drift, I imagine) distributed clocking.  If you find something (check and see if any of the Sonos folks published anything about their clock setup), let us know.  But for now, nothing to do here.

closed 2020-08-10 by earlephilhower