earlephilhower/ESP8266Audio #203: Mono audio option

Adrianotiger, 2019-09-03:
Hi,
I am using the ESP32 with 2 DAC outputs. I would like to use only 1 DAC and not both for the audio.
Is there a way or option to set it only mono?

On the ESP32, pin25 and pin26 are used (DAC1 and DAC2):
https://github.com/earlephilhower/ESP8266Audio/blob/master/src/AudioOutputI2S.cpp#L78
The code is this one:
```
if (output_mode == INTERNAL_DAC || output_mode == INTERNAL_PDM) {
      i2s_set_pin((i2s_port_t)portNo, NULL);
      i2s_set_dac_mode(I2S_DAC_CHANNEL_BOTH_EN);
    } 
```

But if you set the i2s with this:
```
i2s_set_dac_mode(I2S_DAC_CHANNEL_LEFT_EN);
```
Only DAC2 is used and you can use the DAC1 for something else.

But I don't know if the option to set mono can be applied for the other processors too.

thomastech, 2019-09-15:
+1. Thanks for the tip. I'm using a ESP32 and would like to use only one DAC for mono audio. This will free up the other DAC for my project.
- Thomas

nevercast, 2020-06-08:
I believe I saw @bitluni do some neat hacks with I2S and video output at one point, and one of the issues was that you cannot separate the two I2S channels. So you need to pack both streams (LEFT and RIGHT) together.

wmsu2003, 2021-06-16:
Hello,
I changed the AudioOutputI2S.cpp in the way you mentioned.
But I get audio signal on PIN25 and PIN26.
Can someone see an error in my program?

> #include <SPI.h>
> #include <Wire.h>
> 
> long myTimer1 = 0;
> long myTimeout1 = 2000;
> #include <Arduino.h>
> #include "AudioFileSourceSD.h"
> #include "AudioOutputI2S.h"
> #include "AudioGeneratorWAV.h"
> File dir;
> AudioFileSourceSD *source = NULL;
> AudioOutputI2S *out;
> //AudioOutputI2S *out2;
> AudioGeneratorWAV *wav;
> 
> void setup()   {
>   Serial.begin(115200);
>   audioLogger = &Serial;
>   source = new AudioFileSourceSD();
>   out = new AudioOutputI2S(0, 1);
>   out -> SetGain(2.0); //max 4.0
>   wav = new AudioGeneratorWAV();
>   SD.begin();
>   delay(500);           //Abwarten 0,5s
> }
> 
> void loop() {
>   delay(100);
>   wavabspielen1 ("/1756.wav");
> }
> void wavabspielen1 (char Dateiname[]) {
>   File file = SD.open(Dateiname);
>   source->close();
>   source->open(file.name());
>   wav->begin(source, out);
>   while ((wav) && (wav->isRunning())) {
>     if (!wav->loop()) wav->stop();
>   }
> }
> 
> 
>