earlephilhower/ESP8266Audio #204: PCM5102 max power or amplifier needed?

DylanVanAssche, 2019-09-07:
I successfully connected the PCM5102 to an ESP8266 with this library.
With some ear pods, it works fine, but I was wondering if I can connect some 8 ohm 1W speakers to the DAC?

I'm almost sure that I need an amplifier, so I tried to connect a PAM8403 amplifier PCB to it, however nothing happens. I think the PAM8403 PCB is broken since it doesn't work with my PC either. I only connected the left speaker and the audio in for the left speaker to the amplifier.

**Connections**

| PCM5102 | PAM8403 |
| ------------- | ------------ |
| L | L |
| G | Between L and R (GND) |

- Speaker is connected to `+` and `-` of the PAM8403.
- 5V PSU is connected to the PAM8403 power input.

[image]
[image]

connornishijima, 2019-09-13:
Have you tried the right channel yet? It definitely seems that your PAM8403 is faulty, as I've used this exact hookup with success before. Is the PCM5102A still outputting over the line-level jack during this test?

DylanVanAssche, 2019-09-13:
Still output through the jack during the test. I need to try the right channel maybe... I suspect that the PAM8403 is broken...

DylanVanAssche, 2019-09-29:
Speaker was broken! Fixed by replacing the speaker 👍

closed 2019-09-29 by DylanVanAssche