earlephilhower/ESP8266Audio #205: Sound stuttering while doing something else

Mael19, 2019-09-22:
Hello, I'm quite a novice in the Arduino world but I managed to use this awesome library on a ESP32. I still have a smell issue, I'm using a library for a SSD1306 screen (link below) and when I use it (when I make text sliding) the sound is stuttering a bit. It's not that bad but quite annoying... Do any of you have an idea ?

Ticker library : https://github.com/androdlang/InfoTicker

thijsdebont, 2019-09-25:
Stuttering of audio can occur when something else in your code is blocking the continuous running of your program. Think delays, for loops with many cycles or while loops that get temporarily stuck. I'm not familiar with the infoTicker library, but something in it is probably preventing other code to run for a substantial amount of time, causing the audio output to stall.

Mael19, 2019-10-05:
Yeah I know this and I looked into the library without finding any delay... Maybe it's the fact that it has to refresh the screen every 20ms or so. I'll try to check it out more when I get time.

imayoda, 2020-04-15:
@Mael19 do you have i2c screen ? if positive, updating the screen buffer just takes too much time .. I'm in the same situation right now and would be glad to know if you resolved it in some manner..

Mael19, 2020-04-15:
Yeah I have an i2c screen, I thought about the same things. I didn’t manage to do it but I thought about the option of using the dual core capability of the esp32. My idea was to run the webradio part on one core and the display part on the other. Maybe you should give it a try.

ikostoski, 2020-04-15:
@imayoda on ESP32 you have few options, run the decoder->loop() in separate FreeRTOS task (probably the better option) or increase DMA buffers so until it can tolerate longer delays while you update the screen.

imayoda, 2020-04-15:
that's a boomer i'm on the low side :) esp8266 without spiram and even without dac.. haha
I have reached about a click in 5 seconds maxing out the i2c frequency (how to check if effectively 800Khz- 1MHz bitbanging is effectively running ?) and selectively refreshing the screen every 3000-5000 full loops (by writing only the old pixels used in the frame before) ... but still there's a click every 4-5 seconds

ikostoski, 2020-04-15:
@imayoda, can you try to sneak in calls to decoder->loop() somewhere while updating the screen, i.e. update the screen in parts and call decoder->loop() in between ?

imayoda, 2020-04-15:
nice idea, as regular loop is already at it's tiniest, I have to infiltrate the decoder loop() somewhere behind i2c update ... but I do have the feeling that something nasty happens at any Wire activation.

earlephilhower, 2020-05-11:
Looks like this has been explained and the workaround (putting `mp3->loop()` calls through long-running operations described.

closed 2020-05-11 by earlephilhower

Stefan300381, 2024-09-16:
Hi, 

sorry to bring this super old topic back. Can you please explain for dummies what's required?
I stream mp3 via http url and I get a short stutter every time I display something on I2C screen, or whenever I use a connected rotary switch. Is this really solvable via software, or do I need to ad a capacitor somewhere?

thanks in advance

imayoda, 2024-12-29:
> Hi,
>
> sorry to bring this super old topic back. Can you please explain for dummies what's required? I stream mp3 via http url and I get a short stutter every time I display something on I2C screen, [...]

It was resolved in my project using partial screen rewriting and calling mp3->loop() very very often.. .without ram buffer and second core, everything longer than 20-30ms will cause audible disruption in audio stream..