earlephilhower/ESP8266Audio #206: Sometimes audio is distorted

DylanVanAssche, 2019-09-29:
I would like to use this library to play a MP3 file.
When the `A0` pin of the ESP8266 is lower than `100`, the sound loop must be repeated.
Works fine most of the time, but approximately 1 out of 10 times, the audio is distorted (see video). I have this on a breadboard, but also on the PCB you see in the video.

I use an Adafruit Mono amplifier (as advertised in the README).

```
#include <Arduino.h>
#ifdef ESP32
    #include <WiFi.h>
#else
    #include <ESP8266WiFi.h>
#endif

#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

AudioFileSourceSPIFFS *file;
AudioGeneratorMP3 *mp3;
AudioOutputI2S *out;

void setup()
{
  WiFi.mode(WIFI_OFF); 
  Serial.begin(115200);
  delay(1000);
  Serial.printf("MP3 start\n");
  file = new AudioFileSourceSPIFFS("/bell.mp3");
  out = new AudioOutputI2S();
  mp3 = new AudioGeneratorMP3();
  mp3->begin(file, out);
}

void loop()
{
  if(analogRead(A0) < 100 && !mp3->isRunning()) {
    file = new AudioFileSourceSPIFFS("/bell.mp3");
    out = new AudioOutputI2S();
    mp3 = new AudioGeneratorMP3();
    mp3->begin(file, out);
    Serial.printf("MP3 started\n");
  }

  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
  }
}

```

Test project with the MP3 file. 
[PlayMP3Test.zip](https://github.com/earlephilhower/ESP8266Audio/files/3667380/PlayMP3Test.zip)

The issue can been seen in this video:
https://youtu.be/FQ_Q3Eaw39s

DylanVanAssche, 2019-09-29:
Update:

With the Adafruit amplifier (https://www.adafruit.com/product/3006), I can reproduce it. With the PCM5102 + PAM8403 amplifier, this works fine!

However, the Adafruit one is much smaller, it would be handy to be able to use that one.

srihariash999, 2019-10-06:
Can you help me with this, I want to know the connections you made and boards used, I have an ESp8266, a PAM8403 amp, I want to output audio to a speaker.

DylanVanAssche, 2019-10-06:
I connected it like in this issue: https://github.com/earlephilhower/ESP8266Audio/issues/204 between the PAM8403 amp and the PCM5102 DAC.
The PCM5102 DAC is connected like this:

BCK -> ESP8266 NodeMCU RX
DIN -> ESP8266 NodeMCU D8
LCK -> ESP8266 NodeMCU D4

DylanVanAssche, 2020-02-22:
@earlephilhower Do you have any suggestion on how to debug this case?

Additional approaches I tried:
- I tried to use the PROGMEM storage instead of SPIFFS but no luck, the noise is still there.
- Keep looping the file continuously and just `setGain(0.0)` for stopping the sound and `setGain(1.0`) for starting the sound. I thought that maybe by interrupting the generator (`stop()`) before the file ends was the cause maybe, but no luck either.

There's no output to the Serial Monitor of any problems and it goes away when the file is replayed the next time.

jan-cipa, 2020-07-16:
Hi, I had same problem as yours (when using core 2.6.3), today I upgraded to core 2.7.2 and intermittent noise is gone.

earlephilhower, 2020-07-16:
Thanks for the update.  There was a change to the I2S init routine in 2.7.1 which should have cleared this up (holding the clocks in reset until the config is fully set).

closed 2020-07-16 by earlephilhower