earlephilhower/ESP8266Audio #210: my ESP32 play MP3 just output noise

xavierzhu, 2019-10-22:
I use the example:StreamMP3FromHTTP for test.
I only change the url to my mp3 url.
And connect the gpio25 and gnd to the speaker.
But there is only noise.
I changed the MP3 to AAC,and the same.
I test it on esp8266 and OK.
Why ESP32 not OK?

cyakimov, 2019-10-24:
Same problem here. I'm playing a local SPIFFS mp3 and all I get is noise. I tried the `AudioFileSourcePROGMEM` example and it works fine.

> in this house we obey the laws of thermodynamics!

rfestag, 2019-12-16:
It is currently working for me with SPIFFS. I'm specifically using an Adafruit ESP32 feather combined with their Prop Maker wing, and the onboard I2S as opposed to something external (Prop Maker provides an amplifier connected to GPIO#26/A0)

The main difference is that I'm using "AudioOutputI2S" instead of "AudioOutputI2SNoDAC" for my output. Check out this for more info:
https://github.com/earlephilhower/ESP8266Audio#audiooutput-classes. Specifically:

>To use the internal DAC on ESP32, instantiate this class as AudioOutputI2S(0,1), see example PlayMODFromPROGMEMToDAC and code in AudioOutputI2S.cpp for details.

In my case, I'm also setting the output to Mono:

```
  out = new AudioOutputI2S(0, 1);
  out->SetOutputModeMono(true);
```

earlephilhower, 2020-08-08:
The example in question is NOT for a DAC.  It's the `NoDAC` as can be seen at the top of the file.  Change the object to an `AudioOutputI2S` and you're all set, or remove the DAC and use the 1-T circuit shown in the readme.

closed 2020-08-08 by earlephilhower