earlephilhower/ESP8266Audio #211: Use Serial and noDAC?

dagnall53, 2019-11-01:
Is there a way to "disengage" the NoDac drive to RX when I am not playing sounds so I can check for serial inputs? 
I have a variable _soundON that is true if sounds are playing, and I would like to add something in the MAIN Loop call to AudioLoop that would allow me to check Serial inputs.
Any ideas?? 
Many thanks
Dagnall

earlephilhower, 2019-11-01:
It's a hardware limitation, but I'm not sure how limited it is.  The I2S pins have dedicated muxing HW and while it may be possible to disable it w/o stopping I2S, I've never tried it.

Don't forget, it's just one set of pins, so if you have an external serial chip driving at the same time as the I2S is driving, very bad things will happen.

You could disable the GPIO drive (use the actual register write version, not the Arduino pinMode, as that would actually change muxes/etc), then read the GPIO register directly.  How you go about checking for serial comms, which is an async process and needs to always be running, I don't know and doubt it's possible in the general sense.

You could always stop the I2S playback, do whatever serial work, then restart, of course.

Good luck
-EFP3

closed 2019-11-01 by earlephilhower

dagnall53, 2019-11-01:
Thanks..
I think the answer for me is to enable the audio ONLY once I have done any setup (which is done using the Serial port).
Seems to work ok...
Cheers