earlephilhower/ESP8266Audio #213: Software volume control/setting when using "AudioOutputI2S()"

Rob58329, 2019-11-17:
I am using "AudioOutputI2S()" together with "ESP8266SAM" on an ESP8266.  What is the best way to set/reduce the volume in software (a perminant library-mod solution would be fine for me if easiest)?

(FYI: I ask as I am using an Adafruit MAX98375A i2s amplifer, and it is not possible to set the gain of the MAX98375A reliably/consistenly with hardware:   If the ESP8266 reboots/crashes, the MAX98275A (at 3.3v) often seems to reset it's gain to the default 9dB, and ignore the 3dB (100k resitor) setting, requiring the removal of all Vsupply-power which USUALLY gets it to re-read the 100k resistor and change gain to 3dB.    PS: at Vsupply=3.3v with the gain at 9dB, the output is loud and rather distorted - I suspect because this level of output is hitting the 3.3v rail.)

Many thanks for any suggestions.

earlephilhower, 2019-11-17:
There's a SW scaler built-in using fixed-point math so it's fast.  Just add a call to `setGain(0.0-4.0)` (multiplier, not dB level) and the stream's output

Rob58329, 2019-11-18:
@earlephilhower 

Very many thanks for the above info.  (I edited the "AudioOutputI2S.cpp" file and changed "SetGain(1.0);" line to "SetGain(0.1);", and it works a treat!) 👍

I have 3 related questions:

(1) Is it possible to put the "SetGain(0.1)" somewhere in my main sketch (eg. after the "AudioOutputI2S * out = new AudioOutputI2S();" line), and if so what exactly would I need to put?  (OR, is it requried that this edit is in the "AudioOutputI2S.cpp" file?)

(2) Is it possible to view the code used by the "SetGain()" call (ie. which .ccp file is the code in), or is this a hidden/compiled function?

(3) I notice that just adding the "#include <AudioOutputI2S.h>" line in my sketch, even if I don't ever use it (ie. no "AudioOutputI2S * out = new AudioOutputI2S();" or other related commands) still compiles all the "AudioOutputI2S.ccp" stuff, and also all the "ESP8266HTTPClient.h" and associated wifi libraries, and increases my sketch size even if I dont use any of either the ""AudioOutputI2S" or "WiFi" functionaltiy.  Is this normal, and what is the reason?

Finally, one additional note which might be of interest to you:

(4)  I note that using the current "https://github.com/earlephilhower/ESP8266Audio" (v18Nov19), something in the "#include <AudioOutputI2S.h>" uses "SdFat.h" file (the "ESP8266SdFat" library) which seems to have been depreciated in the latest "https://github.com/esp8266/Arduino" (v18Nov19) github.
* Therefore you get the compile error "...\esp8266\libraries\SDFS\src/SDFS.h:36:19: fatal error: SdFat.h: No such file or directory" even if not using any file-systems,
* {ie. to get it to compile I have to manually copy an old verion of the "ESP8266SdFat" library into my directory-structure.)

earlephilhower, 2019-11-18:
1) Set it after you create the AudioOutputI2S object.  out->SetGain(0.2), etc.

2) The SetGain is a member of the AudioOutput class (which AudioOutputI2S subclasses).  The AudioOutput.h file sets it and shows the fixed point multiply in the Amplify() routine.

3) It may compile them, but the linker should not include them in the final binary unless you use those functions/objects.  So they should not take up space in the final executable

4) D'oh.  I'm the guy who rewrote the SDFat stuff in the Arduino Core.  So I busted it...I'll have to go back and clean this library up.  It's just been working for me in my clocks so well that I've never had a reason to go back and recompile...

Rob58329, 2019-11-18:
@earlephilhower 
1&2: Thanks for this - it seems the lowest gain I can set is "out->setGain(1.0/63);" which after truncation gives an impressive 1/64 or -36dB!

3: re. binary file-size:  This is what I would have thought, but for the simple sketch below, just adding the "#include <AudioOutputI2S.h>" line changes the resultant binary file-size from 266,288 bytes to 321,744 bytes (ie. 55,456 bytes MORE), which I don't really undersand!

```
#include <ESP8266WiFi.h>
#include <AudioOutputI2S.h>  // Library"github.com_earlephilhower_ESP8266Audio-master" AND ALSO "github.com_Gianbacchio_ESP8266_Spiram-master"

void setup() {
  Serial.begin(74880);
  Serial.println(F("Start"));
} 

void loop() {}

// (1) With "#include <AudioOutputI2S.h>" COMMENTED-OUT
//.bin file size = 260 KB (266,288 bytes)
//Sketch uses 262128 bytes (25%) of program storage space. Maximum is 1044464 bytes.
//Global variables use 27736 bytes (33%) of dynamic memory, leaving 54184 bytes for local variables. Maximum is 81920 bytes.

// (2) With "#include <AudioOutputI2S.h>" IN
//.bin file size = 314 KB (321,744 bytes)    (ie. 55,456 bytes MORED???)
//Sketch uses 317584 bytes (30%) of program storage space. Maximum is 1044464 bytes.
//Global variables use 28168 bytes (34%) of dynamic memory, leaving 53752 bytes for local variables. Maximum is 81920 bytes.`

```
[PS. this is using the latest Arduino-IDE 1.8.10 and latest Github ESP8266/arduino code (as at 18Nov19)]

earlephilhower, 2020-08-08:
So I've re-checked just now (only 9 months later!) and there is no SD.h included in the library except in `AudioFileSourceSD.h:#include <SD.h>` and there are no instances of SdFat.h, so I can't repro any issues with the code.  Builds seem to run fine on SD and non-SD for me.  Closing as I don't see anything now...

closed 2020-08-08 by earlephilhower

Rob58329, 2020-08-08:
@earlephilhower:
Works fine for me now (with ESP8266 github at 8Aug20), so I'm guessing it may have been a temporary weird error with my SD Library.
Very many thanks!  :-)

WTFredG, 2024-05-24:
Is it possible to change the gain during the playin? I want to make a fade out of it.