earlephilhower/ESP8266Audio #230: Zero i2s dma buffer after write, so there is silence after buffer underflow

Rob58329, 2019-12-01:
@earlephilhower
On all my ESP32 devices (several types). using ("#include <ESP8266SAM.h>" and "#include <AudioOutputI2S.h>"): The ESP32 the i2s Dout signal gives a loud unpleasant buzzing noise as soon as any speech has finished, which if using "AudioOutputI2S *out = new AudioOutputI2S();" won't stop until you issue a "out->stop();" (or start new speech).  (No similar issue with the ESP8266!)

I am guessing this is the issue referred to in "https://github.com/espressif/esp-idf/issues/1789" entitled "Zero i2s dma buffer after write, so we have silence after buffer underflow".  And also related to "https://github.com/earlephilhower/ESP8266Audio/issues/48" which is "esp32: send silence on error?".

I am unclear if any of the suggestions in this #1789 issue are implemented in the current ESP32 or AudioOutputI2S github software versions (as at 18Nov19), as I still get this above unpleasant buzzing noise issue?

The only current way I can get (non-buzzing) multiple (separate) words/sentences, seems to be to issue "delete sam; out->stop();" after every word/sentence, and then start each new word/sentence with "out->begin(); sam=new ESP8266SAM;".

Is this currently the best way to do this?

[ALTHOUGH interestingly I have found that if I end a word/sentance with a full-stop, then this buzzing issue does NOT seem to happen (eg. see the "Hello." word in the code below) - is this a reliable way forward, and is this a documented/un-documented fix or ???]

My ESP32 test code is:

```
#include <WiFi.h>
#include <ESP8266SAM.h>      // Library="github.com_earlephilhower_ESP8266SAM-master"
#include <AudioOutputI2S.h>  // Library"github.com_earlephilhower_ESP8266Audio-master" AND ALSO "github.com_Gianbacchio_ESP8266_Spiram-master"

void setup() {
  Serial.begin(115200); 
  WiFi.mode(WIFI_OFF);
  btStop();
  Serial.println("\nTTS initialisation & speech");
  AudioOutputI2S *out = new AudioOutputI2S();
  out->SetPinout(26, 5, 25); // (clk,lrc,Dout) (any pins eg.16->33)
  
  sam->Say(out,"Hello.");
  delay(3000); // No loud buzzing here because the above ended in "." !??
  sam->Say(out,"Goodbye");
  delay(3000); // But get loud buzzing during this delay()
  delete sam;
  delay(3000); // also get loud buzzing during this delay()
  out->stop(); // buzzing noise stops after this

  delay(3000); // no buzzing noise
  out->begin();
  out->SetGain(0.1); // 0.1 is -20dB
  sam = new ESP8266SAM;
  sam->Say(out,"The End");
  delete sam;
  out->stop(); // No delays before here, so no buzzing noise
}

void loop() {}
```

Any comments regarding the above would be appreciated!

earlephilhower, 2019-12-01:
The buzzing is the ESP32 DMA unit sending out the last buffer it got over and over again.  I thought it sent silence (on the 8266 we do this IIRC) on the 32, but I guess not.

As this is an async process, maybe adding an out->silence() call would be the best way to avoid this everywhere.  All it would need to do is send a block of 0s which would be ignores on the 8266 and looped on the ESP32.  Needs a little more thought.

The "." in SAM probably sends a long period of 0-s, so it does the same thing, effectively.

Rob58329, 2019-12-02:
@earlephilhower
Very many thanks for info/c;larification!  👍 

As it sounds like ensuring each sentance finishes in "." will reliably stop the buzzing on the ESP32 (ie. clear the last buffer), I can use this solution!  Eventually having an "out->silence()" or similar call makes sense, perhaps also if it could be automatically included in the "sam->Say(out,"xxx")" function?!

closed 2019-12-02 by Rob58329