earlephilhower/ESP8266Audio #231: ESP32 Mostly generating noise

Lebo77, 2019-12-18:
I am interfacing my ESP32 with an external MAX98357a breakout board. I have the pins wired up correctly but when I play ab AAC file over I2S (Homer Simpson from PROGMEM)I can hear the audio, but only under a base layer of INTENSE noise. If I apply a lower gain to the program the "signal" goes down but the noise does not. 

The Rudolph the Red-Nosed Reindeer demo that directly synthesizes the audio works perfectly however, so I know it's not the wiring or speaker or DAC. 

Any ideas?

chacalnoir, 2020-01-02:
I was having a similar issue with a slightly different setup.  ESP32 using I2S, internal DAC, PAM8403 Class D amp, and a WAV file read from an SD card.  It turns out the signal from the DAC to the PAM8403 was too large, which, among other issues, caused a high amount of noise that masked the correct signal.  Once I attenuated that signal (with a voltage divider), added a capacitor, and added a high-pass filter for the speaker, the noise went away, leaving the signal.  Note that this issue really only presented itself for a song that was louder and with frequencies that were outside the range of the speaker I was using.  The other tests I ran with simplier sounds appeared reasonable.  Is there a possibility the issue is in the speaker/MAX98357a setup, and there are limitations to that setup that are presenting themselves in this particular test case?

martinberlin, 2020-03-07:
I've a similar issue using a TTGO T5 (v2.4) that has an internal amplifier. 
https://github.com/Xinyuan-LilyGO/T5-Ink-Screen-Series 
I hear the MP3 play using the PlayMP3FromSPIFFS.ino example but is mostly noise and piano melody is barely recognizable

earlephilhower, 2020-08-08:
The root of the problem is the example you're running is not for a DAC, it's using an I2SNoDAC which is the 1-T schematic I've shown.  So, it really does look like noise to the DAC (but if you run it directly through the transistor to a speaker it sounds good!).

You can simply change the `I2SNoDAC` to `I2SDAC` in the code to use a hardware DAC.

closed 2020-08-08 by earlephilhower

nancy-ets, 2021-11-19:
Have you tested with playing the sound only? I mean playing audio without doing anything else. I have loud noise when playing the sound if there are some other interrupts interfering such as displaying on Serial ports, external interupts. 
So make sure that you don't have any of them.