earlephilhower/ESP8266Audio #236: Problem with SPI display ILI9341 when using AudioOutputI2SNoDAC

yigitdev, 2020-02-13:
Hi,
i use a esp8266 nodemcu board and have problems to communicate with a SPI ILI9341 display when using AudioOutputI2SNoDAC.

I have tested the Adafruit_ILI9341 and the TFT_eSPI libs, but non of them worked.
The display is connected as follows:

CS pin is connected to D2 (ESP8266EX GPIO4),
RST pin is connected to D3 (ESP8266EX GPIO0),
D/C pin is connected to D4 (ESP8266EX GPIO2),
SDI (MOSI) pin is connected to D7 (ESP8266EX GPIO13),
SCK pin is connected to D5 (ESP8266EX GPIO14),
VCC and LED are connected to pin 3V3,
GND is connected to pin GND of the NodeMCU board.

Without the AudioOutputI2SNoDAC the display works well.
What can be the problem?

Regards

earlephilhower, 2020-02-13:
The GPIOs used for BCLK and LRCLK are always driven w/I2S, even if you're using I2SNoDac (it's HW related).

Check the pin assignments, I bet they overlap with one of the GPIOs you're using.  Reassigning pins on your HW may be the only way forward.

yigitdev, 2020-02-13:
I checkt my wiring with your external SPI RAM wiring and figure out that you don't use the D4 Pin.
So i change the D/C pin to D1 and it works.

I don't know why the D4 has a problem, i could not found any relation to BCLK and LRCLK.

Thanks

earlephilhower, 2020-02-17:
Thanks for the update.

GPIO2(D4 on your board) is I2SO_WS.  Section 1.5 at https://www.espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf has the full pinout.

closed 2020-02-17 by earlephilhower