earlephilhower/ESP8266Audio #240: Problem with ESP32 (PlayMidiFromSPIFFS example)

observator-x, 2020-02-28:
midi->begin(mid, dac); <-this line seem to cause the following on ESP32:

Starting up...

+0 0x3ffb1f38
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0xfefefefe  PS      : 0x00060f30  A0      : 0x800d4490  A1      : 0x3ffb1eb0  
A2      : 0x00000000  A3      : 0x3ffb1edc  A4      : 0x3ffb90b4  A5      : 0x3ffb90b8  
A6      : 0x00060f20  A7      : 0x00000000  A8      : 0x8014a176  A9      : 0x3ffb8058  
A10     : 0xfefefefe  A11     : 0x3ffb1edc  A12     : 0x00000004  A13     : 0x3ffbc4d0  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000001c  EXCCAUSE: 0x00000014  
EXCVADDR: 0xfefefefc  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x7efefefe:0x3ffb1eb0 0x400d448d:0x3ffb1ed0 0x400d639a:0x3ffb1f60 0x400d1523:0x3ffb1f80 0x400d78c6:0x3ffb1fb0 0x4008b4e9:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9564
ho 0 tail 12 room 4
load:0x40080400,len:6320
entry 0x400806a8

other audio examples (using SPIFFS and I2S) seems to work fine.

Orgelpult, 2020-04-19:
I have exactly the same problem! Is there any solution available?

l3chat, 2020-05-23:
Same here with Lolin32.

earlephilhower, 2020-06-06:
Can anyone post a short MCVE (minimum complete example which crashes for this), or are you saying the example itself barfs?  

Also, if you get a crash could you please decode it using the IDK tool which will translate the hex addresses into actual function names and let me see exactly where it's crashing in the code.

Thx

earlephilhower, 2020-08-08:
Guys, if you get a stack dump on the crash log it would also be useful.  W/O that, there's nothing I can do as I don't run my ESP32 much/at all anymore...

closed 2020-08-08 by earlephilhower