earlephilhower/ESP8266Audio #242: Problem with StreamMP3FromHTTP

DavidMR91, 2020-03-02:
Hello,
I 'm trying the demo "StreamMP3FromHTTP".
I'm using an ESP32 Dev Kit With the MAX98657A I2S Mono Decoder. 
I have changed the AudioOutputI2SNoDAC by AudioOutputI2S

The Code is the following:
`#include <Arduino.h>

#ifdef ESP32
    #include <WiFi.h>
#else
    #include <ESP8266WiFi.h>
#endif
#include "AudioFileSourceICYStream.h"
#include "AudioFileSourceBuffer.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

// To run, set your ESP8266 build to 160MHz, update the SSID info, and upload.

// Enter your WiFi setup here:
const char *SSID = "TESTSSID";
const char *PASSWORD = "TESTPASS";

// Randomly picked URL
const char *URL="http://eurek.org/main/xxMP3test/mp3song.mp3";

AudioGeneratorMP3 *mp3;
AudioFileSourceICYStream *file;
AudioFileSourceBuffer *buff;
AudioOutputI2S *out;

// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);
  (void) isUnicode; // Punt this ball for now
  // Note that the type and string may be in PROGMEM, so copy them to RAM for printf
  char s1[32], s2[64];
  strncpy_P(s1, type, sizeof(s1));
  s1[sizeof(s1)-1]=0;
  strncpy_P(s2, string, sizeof(s2));
  s2[sizeof(s2)-1]=0;
  Serial.printf("METADATA(%s) '%s' = '%s'\n", ptr, s1, s2);
  Serial.flush();
}

// Called when there's a warning or error (like a buffer underflow or decode hiccup)
void StatusCallback(void *cbData, int code, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);
  // Note that the string may be in PROGMEM, so copy it to RAM for printf
  char s1[64];
  strncpy_P(s1, string, sizeof(s1));
  s1[sizeof(s1)-1]=0;
  Serial.printf("STATUS(%s) '%d' = '%s'\n", ptr, code, s1);
  Serial.flush();
}

void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.println("Connecting to WiFi");

  WiFi.disconnect();
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  
  WiFi.begin(SSID, PASSWORD);
[... 647 lines cut ...]
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 34781'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 34782'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 34783'
STATUS(mp3) '565' = 'Decoding error 'bad main_data_begin pointer' at byte offset 347'
Running for 9555 ms...
Running for 10556 ms...
Running for 11557 ms...
Running for 12558 ms...
Running for 13559 ms...
Running for 14560 ms...
Running for 15561 ms...
Running for 16562 ms...
STATUS(buffer) '3' = 'Buffer underflow'
STATUS(buffer) '2' = 'Refilling buffer'
STATUS(buffer) '3' = 'Buffer underflow'
MP3 stop, len==0
MP3 done
MP3 done
MP3 done
MP3 done
MP3 done
MP3 done
MP3 done
MP3 done
`

DerMega, 2020-03-09:
I have got the same problem and never got an answer or idea :-/
Unfortunately this is the only pure "Music Stream Class", which is not baked in to a complete radio script, I found after hours of searching.

BastiG, 2020-04-24:
For me this was primarily related to embedded ID3 tags (not sure if it's particularly due to ID3v2 which is "part" of the audio stream, syncsignal magic).
Both of these solutions worked for me:
- Strip mp3 files of all ID3 tags
- Replace the AudioGeneratorMP3 with AudioGeneratorMP3a which uses the Helix MP3 decoder

Avon1, 2020-06-05:
Sorry I did not understand what do you mean by remove 

- "Strip mp3 files of all ID3 tags"

I did replace  AudioGeneratorMP3 with AudioGeneratorMP3a it did not work.

I am streaming with a local url with this software,
- http://www.streamwhatyouhear.com/?source=swyh

Example of local stream from computer sound

- "http://192.168.1.13:8000/stream/swyh.mp3"

mxk-ru, 2021-05-04:
Me too ((
Some body can help to solve that problem?

kingofnull, 2021-06-23:
Try to set out to mono and fix samplerate. Also make sure have stable power supply , lose connections and poor breadboard contacts cause to brownouts and memory corruption. Also you should have good wifi signal its very important in sync problem. It's my code and it's stable often :

```
#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#else
#include <ESP8266WiFi.h>
#endif
#include "AudioFileSourceICYStream.h"
#include "AudioFileSourceBuffer.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"
// To run, set your ESP8266 build to 160MHz, update the SSID info, and upload.
// Enter your WiFi setup here:
const char *SSID = "_";
const char *PASSWORD = "DarkNight";
// Uncomment one link (I have added 6 radio streaming link, you can check each)
//flawlessly working radio streaming link
//const char *URL="http://ndr-edge-206c.fra-lg.cdn.addradio.net/ndr/njoy/live/mp3/128/stream.mp3"; //'N-JOY vom NDR - www.n-joy.de'
const char *URL = "http://0n-80s.radionetz.de:8000/0n-70s.mp3";
//const char *URL="http://jazz.streamr.ru/jazz-64.mp3";
// It will work but buffer alot
//const char *URL="http://stream.ca.morow.com:8003/morow_med.mp3";
//const char *URL= "http://ndr-ndr1radiomv-schwerin.sslcast.addradio.de/ndr/ndr1radiomv/schwerin/mp3/128/stream.mp3";
//const char *URL="http://mms.hoerradar.de:8000/rst128k";//Radio RST(German)
AudioGeneratorMP3 *mp3;
AudioFileSourceICYStream *file;
AudioFileSourceBuffer *buff;
AudioOutputI2SNoDAC *out;
// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);
  (void) isUnicode; // Punt this ball for now
  // Note that the type and string may be in PROGMEM, so copy them to RAM for printf
  char s1[32], s2[64];
  strncpy_P(s1, type, sizeof(s1));
  s1[sizeof(s1) - 1] = 0;
  strncpy_P(s2, string, sizeof(s2));
  s2[sizeof(s2) - 1] = 0;
  Serial.printf("METADATA(%s) '%s' = '%s'\n", ptr, s1, s2);
  Serial.flush();
}

void initA() {
  file = new AudioFileSourceICYStream(URL);
  file->RegisterMetadataCB(MDCallback, (void*)"ICY");
  buff = new AudioFileSourceBuffer(file, 2048);
  buff->RegisterStatusCB(StatusCallback, (void*)"buffer");
  out = new AudioOutputI2SNoDAC();
  out->SetOutputModeMono(true);
  out->SetGain(1);
  out->SetRate(48000);

  mp3 = new AudioGeneratorMP3();
  mp3->RegisterStatusCB(StatusCallback, (void*)"mp3");
  mp3->begin(buff, out);
}

// Called when there's a warning or error (like a buffer underflow or decode hiccup)
void StatusCallback(void *cbData, int code, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);
  // Note that the string may be in PROGMEM, so copy it to RAM for printf
  char s1[64];
  strncpy_P(s1, string, sizeof(s1));
  s1[sizeof(s1) - 1] = 0;
  Serial.printf("STATUS(%s) '%d' = '%s'\n", ptr, code, s1);
  Serial.flush();
}

void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.println("Connecting to WiFi");
  WiFi.disconnect();
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  WiFi.begin(SSID, PASSWORD);
  // Try forever
  while (WiFi.status() != WL_CONNECTED) {
    Serial.println("...Connecting to WiFi");
    delay(1000);
  }
  Serial.println("Connected");
  audioLogger = &Serial;
  initA();
}
void loop()
{
  static int lastms = 0;
  if (mp3->isRunning()) {
    if (millis() - lastms > 1000) {
      lastms = millis();
      Serial.printf("Running for %d ms...\n", lastms);
      Serial.flush();
    }
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
    initA();
  }
}

```

SaitamaTechno, 2024-04-13:
Okay I think I found the solution here. it is all about mp3 sample rate and the secret is in mp3 file types. You just need to find best mp3 file. 
You should change the url as it is here:

const char *URL = "http://0n-80s.radionetz.de:8000/0n-70s.mp3";
// the default url did not work properly, but this one works just fine.
and it works perfectly on esp8266, with only transistor speaker.

lamensje, 2024-07-10:
I had the exact same error. There was nothing wrong with the code nor the ESP8266.
I used nginx as a reverse proxy in front of my Icecast2 server, which does wierd things with buffering.
Using the Icecast2 url directly solved all my stuttering issues.