earlephilhower/ESP8266Audio #243: Overheating problem with 2N3904 NPN using NoDac

yigitdev, 2020-03-04:
i use a esp8266 nodemcu board with a 2N3904 NPN connected to a small speaker 7Ohm 1Watt, it works but i have problems with overheating.

I did the wiring according to your instructions.
What could be the problem?

Regards

earlephilhower, 2020-03-04:
Please try adding 2.2K ohm current-limiting resistor from the ESP8266 I2SOUT and the transistor Base.

Unfortunately, depending on the transistor and your power supplies, it could end up trying to draw a significant amount of current from the output pin and you'd wind up having issues with the 8266 and the transistor itself will start dumping lots of power across the EC path.

yigitdev, 2020-03-07:
i have add 2.2k ohm resistor, now the overheating is gone but now the volume level is lower.

How can i hold the transistor cool and get higher volume? 

Thanks

d-a-v, 2020-03-07:
@earlephilhower As being no electronician,
I understand that the transistor is either shut off or saturated,
I wonder why we can't use a n-mos transistor instead of the bipolar 2n3904 npn one ?

earlephilhower, 2020-03-08:
@yigitdev, yes the lower heating and the lower volume go together.  In this simple circuit, the volume is going to be ~proportional to the base current and transistor beta.  If you have no resistor, the base current is going to be significantly higher, and hence the current EC will be higher, and you'll be dissipating more power (I * Rec) in the transistor, causing the heat.

You can use a lower value resistor, or a potentiometer to adjust and find a sweet spot of temperature vs. volume.

@d-a-v, a NFET does work but gives a worse result.  RDS for a 2N7000 = ~5ohm, and your load is only 8ohm, so you'd end up with a voltage divider at best.  Plus, since the VD=5V, but the VG1=3.3V you're not even full on.

SPICE with NFET 2N7000 shows a swing of only around 0.4V thru the speaker.

The same circuit with an NPN 2N3904 gives 0.9V thru the speaker.  A 2N2222 with a higher beta gives 1.5V thru the speaker.  We don't care about linearity, we're just bang-bang'ing, so actually I should probably update the schematic to suggest the 2N2222 and the 2.2K resistor.

The SPICE is available online at https://www.partsim.com/simulator/#269234

earlephilhower, 2020-05-11:
Closing as the readme has been updated.  The 2N3904 in a TO92 plastic case isn't really spec'd to handle more than a few 10s of mW.  Something larger in a metal can would be appropriate if you need both volume and 1-T.  Also, making sure the I2S pin is driven to disable the transistor when not playing audio will help, but is beyond this library's scope.

closed 2020-05-11 by earlephilhower