earlephilhower/ESP8266Audio #244: using buffer reading mp3 from SPIFFS

tomtobback, 2020-03-09:
Hi, i read the buffer feature is meant for http streaming but i was hoping to use it for reading a file from SPIFFS, as i have another task in my main loop that takes about 50-100ms (decode jpeg and display on tft). However, it does not work as expected, i get stuttering bits of audio at a very low speed. If i comment out the other task (display) it works fine. 
my setup():
```
 audioLogger = &Serial;
  file = new AudioFileSourceSPIFFS("/pno-cs.mp3");
  buff = new AudioFileSourceBuffer(file, 2048);
  buff->RegisterStatusCB(StatusCallback, (void*)"buffer");
  out = new AudioOutputI2S(0);
  out -> SetPinout(27, 26, 25);
  out -> SetGain(0.1);
  mp3 = new AudioGeneratorMP3();
  mp3->begin(buff, out);
```
the audio block in my main loop():
```
    if (mp3->isRunning()) {
      Serial.println(buff->getFillLevel());
      if (!mp3->loop()) mp3->stop();
      Serial.println(buff->getFillLevel());
    } else {
      Serial.printf("MP3 done\n");
      delay(1000);
    }
```
The callback function is triggered only once, at the very first loop and return ```STATUS(buffer) '2' = 'Refilling buffer'``` 
I'm printing the FillLevel of the buffer and it shows 2047 all the time, so the buffer does not seem to be used?
Any advice for such case?

ikostoski, 2020-03-09:
Hi tomtobback, if running on ESP32, instead of using input buffer, try increasing the DMA output buffer, the third parameter of AudioOutputI2S, i.e. `out = new AudioOutputI2S(0, 0, 32);`  and see if that helps.

tomtobback, 2020-03-10:
@ikostoski thanks much, that indeed works, without using the AudioFileSourceBuffer()
i had to increase the dma_buf_count to 128 to have smooth sound. Increasing it further to 256 gives panic reboots. Just for reference, my configuration below, which allows up to 100ms between mp3->loop() calls. Thanks again!
```
  file = new AudioFileSourceSPIFFS("/pno-cs.mp3");
  out = new AudioOutputI2S(0, 0, 128);
  out -> SetPinout(27, 26, 25);
  out -> SetGain(0.1);
  mp3 = new AudioGeneratorMP3();
  mp3->begin(file, out);
```

earlephilhower, 2020-05-11:
The root cause of this would be spending too long between `mp3->loop()` calls, which would cause smaller DMA buffers to underflow.  Closing since you seem to have a brute-force way of increasing the DMA size.

Alternatively, sprinkling `mp3->loop()` calls throughout long-running sections of code would also work fine.

closed 2020-05-11 by earlephilhower