earlephilhower/ESP8266Audio #246: Crackles if stopping mp3 mid-playback

thijsdebont, 2020-03-26:
First of all, thanks so much for this library. Really awesome. I'm working on a keyboard-like project. Press a key and a sample will play. I have a question about stopping a sample (mp3) mid-playback and starting another. If I start a mp3 and let it play out, everything is perfect. However, if I stop playback while the mp3 is playing, there's an audible short crackle most of the times. Strange thing is, this happens at the start of playback too sometimes, if the previous sample didn't play out entirely. 

I'm using I2S playback with an MAX98357A DAC (same as adafruit) connected to an ESP32.  MP3's are coming from a SD card. I brought the sketch (see below) I'm using down to a bare minimum of one button used to trigger one sample. Now, I've read some things in the other issues about DMA, buffers, objects needing to be deleted etc, but nothing seems to help. Deleting the file and/or mp3 object crashes my ESP. Any help is appreciated.

`#include "driver/rtc_io.h"
#include <Preferences.h>              // Store values in NV ram (instead of EEPROM, which is depricated on ESP32)
#include <Bounce2.h>

#include "AudioFileSourceSD.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

#define POWER_SD  15
#define POWER_5V  17
#define PWR_BTN   35

// Audio variables
volatile bool playing = 0;
float fCurrentVolume  = 80;
float fMainVolume     = 80;

// Connection flags and timers
unsigned long lMqttConnRetryTimer = -5000;
bool bConnected = false;
bool bWebserver = false;

// Power flags and timers
unsigned long GET_BATT_TIMER  = 60000;          // Time between battery voltage measurement
unsigned long lPrevGetBatt = millis();          // Timer to keep track of last battery measurement

// Files and folders
unsigned int nFolders = 0;
unsigned int nFiles   = 0;

AudioGeneratorMP3 *mp3;
AudioFileSourceSD *file;
AudioOutputI2S *out;

Bounce debouncer = Bounce();                                      // Tap button debouncer

Preferences pref;

void setup() {
  pinMode(PWR_BTN, INPUT_PULLDOWN);
  pinMode(POWER_5V, OUTPUT);
  pinMode(POWER_SD, OUTPUT);
  digitalWrite(POWER_5V, HIGH);         // Apply power to 5V supply (LED's and amplifier)
  
  Serial.begin(115200);
  delay(50);
  
  digitalWrite(POWER_SD, HIGH);
  delay(200);                                       // Wait for SD card to initialize
  SD.begin();                                       // Start SD card

  debouncer.attach(PWR_BTN);
  debouncer.interval(25);
  
  out = new AudioOutputI2S(0,0,32,-1);
  file = new AudioFileSourceSD();
  mp3 = new AudioGeneratorMP3();
  
  out->SetGain(0.5);
  
}

void loop() {
  fnHandleButton();

  if(mp3->isRunning()) {
    if (!mp3->loop()) {
      mp3->stop();
      file->close();
    }
  }
}

void fnPlay (String stFile) {
  if(mp3->isRunning()) {
    mp3->stop();
    file->close();
  }
  
  if(stFile.indexOf("/") != 0) {
    stFile = "/" + stFile;
  }
  
  file->open((char*)stFile.c_str());
  mp3->begin(file, out);
}

void fnStop() {
  if(mp3->isRunning()) {
    mp3->stop();
    file->close();
  }
}

void fnVolume (float fVolume) {
  fCurrentVolume = fVolume;
  float fGain = (fCurrentVolume / 100) * (fMainVolume / 100);
  out->SetGain(fGain);
}

void fnHandleButton() {
  debouncer.update();
  if(debouncer.rose()) {
    fnPlay("rhodes/c2_rhodes.mp3");
  }
  if(debouncer.fell()) {
    fnStop();
  }
}`

thijsdebont, 2020-03-30:
Answering my own question :). The problem with the tick at the start of playback seemed to be caused by not properly deleting the mp3 and file objects. I know there's word about updating the examples to better reflect this. That would have helped in this case...

The ticks at the end of the samples are caused by audio being cut off sharply if stopping mid-playback. More info in new issue/feature request #248

jjbboox, 2020-05-05:
you can set mute pin of dac before stop.

Dawnlord, 2023-02-01:
I meet the same issue. If the last mp3 is stopped at mid. It will have pop. Then, there will also be pop at the beginning of the next mp3.
Do you have the solution for this issue now?