earlephilhower/ESP8266Audio #253: VSCode plateformio error

ririfonfon, 2020-04-15:
`In file included from .pio\libdeps\hw2-kxkm\ESP8266Audio\src\AudioFileSourceSPIRAMBuffer.h:27:0,
                 from .pio\libdeps\hw2-kxkm\ESP8266Audio\src\AudioFileSourceSPIRAMBuffer.cpp:26:
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:46:38: error: 'SPI1CMD' was not declared in this scope
         spi_regs *spi1 = (spi_regs*)&SPI1CMD;
                                      ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h: In member function 'void ESP8266SPIRAM::spi_init()':
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:70:13: error: 'GPMUX' was not declared in this scope
             GPMUX &= ~(1 << 9);
             ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h: In member function 'void ESP8266SPIRAM::spi_writetransaction(int, int, int, int, ESP8266SPIRAM::iotype)':
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:84:36: error: 'SPIBUSY' was not declared in this scope
             while (spi1->spi_cmd & SPIBUSY) { delay(0); }
                                    ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:87:42: error: 'SPIUADDR' was not declared in this scope
             spi1->spi_user = (addr_bits? SPIUADDR : 0) | (dummy_bits ? SPIUDUMMY : 0) | (data_bits ? SPIUMOSI : 0) | (dual ? SPIUFWDIO : 0);
                                          ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:87:72: error: 'SPIUDUMMY' was not declared in this scope
             spi1->spi_user = (addr_bits? SPIUADDR : 0) | (dummy_bits ? SPIUDUMMY : 0) | (data_bits ? SPIUMOSI : 0) | (dual ? SPIUFWDIO : 0);
                                                                        ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:87:102: error: 'SPIUMOSI' was not declared in this scope
             spi1->spi_user = (addr_bits? SPIUADDR : 0) | (dummy_bits ? SPIUDUMMY : 0) | (data_bits ? SPIUMOSI : 0) | (dual ? SPIUFWDIO : 0);
                                                                                                      ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:87:126: error: 'SPIUFWDIO' was not declared in this scope
             spi1->spi_user = (addr_bits? SPIUADDR : 0) | (dummy_bits ? SPIUDUMMY : 0) | (data_bits ? SPIUMOSI : 0) | (dual ? SPIUFWDIO : 0);
                                                                                                                              ^
In file included from .pio\libdeps\hw2-kxkm\ESP8266Audio\src\AudioFileSourceSPIRAMBuffer.h:27:0,
                 from .pio\libdeps\hw2-kxkm\ESP8266Audio\src\AudioFileSourceSPIRAMBuffer.cpp:26:
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:94:33: error: 'SPIBUSY' was not declared in this scope
                 spi1->spi_cmd = SPIBUSY;
                                 ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:98:33: error: 'SPIBUSY' was not declared in this scope
                 spi1->spi_cmd = SPIBUSY;
                                 ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h: In member function 'uint32_t ESP8266SPIRAM::spi_readtransaction(int, int, int, int, ESP8266SPIRAM::iotype)':
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:106:36: error: 'SPIBUSY' was not declared in this scope
             while (spi1->spi_cmd & SPIBUSY) { delay(0); }
                                    ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:109:42: error: 'SPIUADDR' was not declared in this scope
             spi1->spi_user = (addr_bits? SPIUADDR : 0) | (dummy_bits ? SPIUDUMMY : 0) | SPIUMISO | (dual ? SPIUFWDIO : 0);
                                          ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:109:72: error: 'SPIUDUMMY' was not declared in this scope
             spi1->spi_user = (addr_bits? SPIUADDR : 0) | (dummy_bits ? SPIUDUMMY : 0) | SPIUMISO | (dual ? SPIUFWDIO : 0);
                                                                        ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:109:89: error: 'SPIUMISO' was not declared in this scope
             spi1->spi_user = (addr_bits? SPIUADDR : 0) | (dummy_bits ? SPIUDUMMY : 0) | SPIUMISO | (dual ? SPIUFWDIO : 0);
                                                                                         ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:109:108: error: 'SPIUFWDIO' was not declared in this scope
             spi1->spi_user = (addr_bits? SPIUADDR : 0) | (dummy_bits ? SPIUDUMMY : 0) | SPIUMISO | (dual ? SPIUFWDIO : 0);
                                                                                                            ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:117:29: error: 'SPIBUSY' was not declared in this scope
             spi1->spi_cmd = SPIBUSY;
                             ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h: In member function 'void ESP8266SPIRAM::writeBytes(uint32_t, const void*, int)':
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:164:40: error: 'SPIBUSY' was not declared in this scope
                 while (spi1->spi_cmd & SPIBUSY) { delay(0); }
                                        ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h: In member function 'void ESP8266SPIRAM::begin(int, int)':
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:224:35: error: 'SPICDIO' was not declared in this scope
                 spi1->spi_ctrl |= SPICDIO | SPICFASTRD;
                                   ^
.pio\libdeps\hw2-kxkm\ESP8266Audio\src\spiram-fast.h:224:45: error: 'SPICFASTRD' was not declared in this scope
                 spi1->spi_ctrl |= SPICDIO | SPICFASTRD;
                                             ^
*** [.pio\build\hw2-kxkm\lib2d8\ESP8266Audio\AudioFileSourceSPIRAMBuffer.cpp.o] Error 1`

Maigre, 2020-04-15:
Hi,
same issue here, the internal SPIRAM addition broke the compilation for ESP32:

`spiram-fast.h:46:38: error: 'SPI1CMD' was not declared in this scope`

same with GPMUX, SPIBUSY, SPIUADDR, etc..
I guess it is some ESP8266 specifics not available on ESP32 ?

Any include missing to support ESP32 ? 
I'll dig into it and come back if i found something.

Best,

earlephilhower, 2020-04-15:
Sorry, I didn't think it was used on the ESP32 at all (since you've got so much more RAM).

I'll throw a PR in to use the native SPI class on the ESP32.

Maigre, 2020-04-15:
We do not use the spiram, but it is compiled since it's included. Indeed best should be a conditional include if ESP866 is detected.
Thanks!

PR #254 (2020-04-15): Fix ESP32 compilation, add SPI class RAM wrapper
Seems like VSCODE and others try and compile all files in a directory,
even if they aren't used.  Because the spiram-fast had ESP8266-only
registers, it caused an error when attempting to build anything for the
ESP32.

Update by writing a new SPI class based version of the spiram-fast class
(tested on the ESP8266) which uses the SPI wrapper class.

Fixes #253

@Maigre, can you give this a test and make sure it fixes the issue?  Thx!
commit 5b8ba168 (2020-04-15): Fix ESP32 compilation, add SPI class RAM wrapper (#254)

Seems like VSCODE and others try and compile all files in a directory,
even if they aren't used.  Because the spiram-fast had ESP8266-only
registers, it caused an error when attempting to build anything for the
ESP32.

Update by writing a new SPI class based version of the spiram-fast class
(tested on the ESP8266) which uses the SPI wrapper class.

Fixes #253
closed 2020-04-15 by earlephilhower