earlephilhower/ESP8266Audio #260: Playing single midinotes instead of whole midifile?

Orgelpult, 2020-04-18:
Hi!
I wonder how it is possible to play single midinotes instead of a complete midi-file.
I would be really glad, if you could provide an example code for that.
THANK you so much for your help!
Best regards
Michael from Austria

earlephilhower, 2020-04-18:
There is unfortunately not an easy way of doing this.  The tsf.h library does this, and the MIDI audio generator uses it.  You can look at the function `PlayMIDI`
https://github.com/earlephilhower/ESP8266Audio/blob/1ca965b474783df60c9158953fca1b5aebf461d9/src/AudioGeneratorMIDI.cpp#L364
and the `loop` method
https://github.com/earlephilhower/ESP8266Audio/blob/1ca965b474783df60c9158953fca1b5aebf461d9/src/AudioGeneratorMIDI.cpp#L534
to get an idea of the process.

If you can encapsulate the single note as a MIDI file, that may be the simplest way to make it happen.  50 bytes or so constant, with a single note specifier byte(s) that you can change on-the-fly and send to the midigenerator.

Good luck!

Orgelpult, 2020-04-18:
Thank you for your answer. But this would not help much because I have to trigger several midinotes at the same time. So I have to have access to a function like "midi_note_on (midichannel, note, velocity)" and midi_note_off(midichannel, note).
Unfortunately I do not know how to access the existing functions of tsf.h in my arduino code. Could you please take a look on the tsf.h and please tell my how to get access to it in the arduino code?
I would be very thankful for your help!
Best regards
Michael

earlephilhower, 2020-05-11:
Sorry, that's just not supported with this library.  You can look at the TSF upstream docs and see if you can adapt something from them.  Good luck.

closed 2020-05-11 by earlephilhower