earlephilhower/ESP8266Audio #262: MP3 missing final second

geoffday67, 2020-04-21:
When playing an MP3 the final second or so is missing. This seems to be because the `Input()` method in `AudioGeneratorMP3.cpp` returns `MAD_FLOW_STOP` as soon as the file is completely read, which doesn't allow the final MP3 data to be processed. I fixed it by setting a flag to indicate that there is no more file data but not actually returning `MAD_FLOW_STOP` until all the data has been processed. I'm not sure if I've done this in a reliable way though. Here's my altered code:
```
enum mad_flow AudioGeneratorMP3::Input()
{
  int unused = 0;

  if (stream->next_frame) {
    unused = buffLen - (stream->next_frame - buff);
    memmove(buff, stream->next_frame, unused);
    stream->next_frame = NULL;
  }

  if (finishing && (unused == buffLen - 1)) {
    return MAD_FLOW_STOP;
  }

  if (unused == buffLen) {
    // Something wicked this way came, throw it all out and try again
    unused = 0;
  }

  lastReadPos = file->getPos() - unused;
  int len = buffLen - unused;
  len = file->read(buff + unused, len);
  if (len == 0) {
    finishing = true;
  }

  mad_stream_buffer(stream, buff, len + unused);

  return MAD_FLOW_CONTINUE;
}
```
Issues 178 and 179 have the same root cause I'd guess.

earlephilhower, 2020-04-22:
Thanks!  I think you've got it, @geoffday67 

Would you like to do a PR here?

I also saw your fix in the Arduino core, good work.  PRs would get your name in the commit logs for all to see (or blame, as the case may be).

geoffday67, 2020-04-23:
Thanks, I'll do PRs for both.

earlephilhower, 2020-04-25:
@geoffday67 ping.  We're close to freezing 2.7.0 on the core, so if you can get it this weekend (at least the I2S reset patch) that would be great and ensure it makes it in.

geoffday67, 2020-04-26:
Both PRs done, I've not done PRs into a public repo before so hopefully I did it right via a fork. I don't have the setup to build the libs locally afaik but I guess the CI will test it out.

earlephilhower, 2020-04-26:
Thanks!  They were done fine in both repos.

Closing as the change is now merged.

closed 2020-04-26 by earlephilhower