earlephilhower/ESP8266Audio #264: Can support SDMMC

jjbboox, 2020-04-30:


jjbboox, 2020-04-30:
AudioFileSourceSDMMC.h
`#ifndef _AUDIOFILESOURCESDMMC_H
#define _AUDIOFILESOURCESDMMC_H

#include "AudioFileSource.h"
#include <SD_MMC.h>

class AudioFileSourceSDMMC : public AudioFileSource
{
  public:
    AudioFileSourceSDMMC();
    AudioFileSourceSDMMC(const char *filename);
    virtual ~AudioFileSourceSDMMC() override;
    
    virtual bool open(const char *filename) override;
    virtual uint32_t read(void *data, uint32_t len) override;
    virtual bool seek(int32_t pos, int dir) override;
    virtual bool close() override;
    virtual bool isOpen() override;
    virtual uint32_t getSize() override;
    virtual uint32_t getPos() override;

  private:
    File f;
};

#endif
`
AudioFileSourceSDMMC.cpp
`
#include "AudioFileSourceSDMMC.h"

AudioFileSourceSDMMC::AudioFileSourceSDMMC()
{
}

AudioFileSourceSDMMC::AudioFileSourceSDMMC(const char *filename)
{
  open(filename);
}

bool AudioFileSourceSDMMC::open(const char *filename)
{
  f = SD_MMC.open(filename, FILE_READ);
  return f;
}

AudioFileSourceSDMMC::~AudioFileSourceSDMMC()
{
  if (f) f.close();
}

uint32_t AudioFileSourceSDMMC::read(void *data, uint32_t len)
{
  return f.read(reinterpret_cast<uint8_t*>(data), len);
}

bool AudioFileSourceSDMMC::seek(int32_t pos, int dir)
{
  if (!f) return false;
  if (dir==SEEK_SET) return f.seek(pos);
  else if (dir==SEEK_CUR) return f.seek(f.position() + pos);
  else if (dir==SEEK_END) return f.seek(f.size() + pos);
  return false;
}

bool AudioFileSourceSDMMC::close()
{
  f.close();
  return true;
}

bool AudioFileSourceSDMMC::isOpen()
{
  return f?true:false;
}

uint32_t AudioFileSourceSDMMC::getSize()
{
  if (!f) return 0;
  return f.size();
}

uint32_t AudioFileSourceSDMMC::getPos()
{
  if (!f) return 0;
  return f.position();
}

`

earlephilhower, 2020-04-30:
Actually, there's already an `AudioFileSourceFS` object, just pass in `&SD_MMC` to it and, if it supports the Arduino core `File` API, it will "just work."

jjbboox, 2020-04-30:
Thanks for your answer

closed 2020-04-30 by jjbboox