earlephilhower/ESP8266Audio #265: Wait after reading MP3 file for final frames (#263)

NikolaiRadke, 2020-05-04:
Dear Earle,

the new commit is not running properly in my setting with ESP32. Function isRunning() is always returning TRUE after finishing playing the file.

Kind regarts,
Nikolai

earlephilhower, 2020-05-04:
Could you please attach a small sample MP3 that shows this?  On tests here I don't seem to be hitting the case.  There might be some encoding format/length/etc. differences, so a sample would be very handy.

NikolaiRadke, 2020-05-04:
Hi!
Sure. It is a simple "Beep". It runs at startup. I removed it and played some other random files.  After some flawlessly running files, it stopped playing again, so i reckon that your are right, it seems to depend on the encoding format/whatever.

After "Beep", isRunning() always returns TRUE.
Sorry for the Zip, GitHub doesn't allow MP3.

[beep.zip](https://github.com/earlephilhower/ESP8266Audio/files/4576907/beep.zip)

earlephilhower, 2020-05-06:
Got it.  Thanks, @NikolaiRadke.  Let me run it through the hardware this weekend and see if I can figure out a way to make both things work properly.

countrysideboy, 2020-05-08:
Hi, I found the same problem with ESP8266.

countrysideboy, 2020-05-09:
@earlephilhower 
I try to modify the code
from
```
  if (finishing && (unused == buffLen - 1)) {
    return MAD_FLOW_STOP;
  }
```
to
```
  if (finishing && (unused <= buffLen - 1)) {
    return MAD_FLOW_STOP;
  }
```
I didn't go into the code logic, but it seems that I can play music continuously now.

PR #266 (2020-05-11): Redo the last packet handling of the MP3 decoder
Keep track of how many values are actually good in the buffer (before it
was always guaranteed to be the full size) and when EOF is detected,
only shift out valid, good bytes, and return STOP when the unused bytes
are 0 or unchanged from the last pass.

Fixes #265
commit b7cb0ee8 (2020-05-11): Redo the last packet handling of the MP3 decoder (#266)

Keep track of how many values are actually good in the buffer (before it
was always guaranteed to be the full size) and when EOF is detected,
only shift out valid, good bytes, and return STOP when the unused bytes
are 0 or unchanged from the last pass.

Fixes #265
closed 2020-05-11 by earlephilhower