earlephilhower/ESP8266Audio #267: ESP8266 SD card mp3 source sputtering

seesoe, 2020-05-11:
Hello, I'm having trouble getting mp3 files to play properly from sd card. The playback sputters and has occasional pops. I've tried various things with no avail.

`AudioGeneratorMP3a` (worse)
`AudioFileSourceBuffer` (no change)
`AudioOutputI2S` `dma` to `128` (no change)

```c
#include <Arduino.h>

#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceSD.h"
#include "AudioFileSourceBuffer.h"

#include <SD.h>
#include <SPI.h>
#include <FS.h>

AudioGeneratorMP3 *mp3;
AudioFileSourceSD *file;
AudioFileSourceBuffer *buff;
AudioOutputI2S *out;

File root;

void setup() {
  Serial.begin(115200);
  delay(1000);

  Serial.print("Initializing SD card...");
  if (!SD.begin(4)) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");

  audioLogger = &Serial;

  file = new AudioFileSourceSD("/test.mp3");
  buff = new AudioFileSourceBuffer(file, 2048);
  out = new AudioOutputI2S(0, 0, 128);
  out->SetGain(0.02);
  mp3 = new AudioGeneratorMP3();
  mp3->begin(buff, out);
}

void loop() {
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}
```

I tried the loop block found here to see if the buffer was being used and the value didn't change.
https://github.com/earlephilhower/ESP8266Audio/issues/244#issue-577792862

earlephilhower, 2020-05-11:
The symptoms you describe indicate that it's taking too long between calls to mp3->loop().  Try increasing the SPI speed as you may be running it at a much lower rate than it needs to be.

closed 2020-05-11 by earlephilhower

seesoe, 2020-05-11:
I have cpu clock speed set to 160mhz and all the configs to their defaults

I tried these two things with no luck

  `SPI.setFrequency(80000000);`

  `if (!SD.begin(4, SPI_FULL_SPEED)) {`

a442509097, 2022-05-28:
I changed the SD card or changed to SPI_DIV3_SPEED, and it worked