earlephilhower/ESP8266Audio #270: after chaning DMA buffer size, mp3/wav->loop() doesn't play the complete file

don41382, 2020-05-14:
Changing the dma buffer size of `AudioOutputI2S` to a higher value, leads to not completely played songs.

I have created a sample project to showcase the error.
https://github.com/don41382/esp8266-audio-cut-off-demo/blob/master/src/main.cpp

**Play complete**
The first output is created with a buffer of 2 (`out = new AudioOutputI2S(0,0, 2);`) This will play the complete song and you will hear the BEEP at the end of the file.

**Play incomplete**
In the second play, the buffer is set to 16 and you will not hear a BEEP at the end.

It assume the dma is never flushed, after the file has reached it end. 

Any ideas for this bug?

earlephilhower, 2020-05-14:
Could well be.  I haven't used the ESP32 in a while, so I think it's something specific on that size. A PR would be most welcome.

The line that clears the DMA buffers is here:
https://github.com/earlephilhower/ESP8266Audio/blob/e0d413622c829b06186684e3d61e8e2ffa0953f4/src/AudioOutputI2S.cpp#L204

Does the EPS32 SDK have a call that lets you check the level in the DMA buffers?  You could then wait for them to empty and then zero them out at that line, instead of just wiping them immediately.

ikostoski, 2020-05-14:
Unfortunately there is no such API call in ESP-IDF. Also, the underlying FreeRTOS queue object (which holds the buffer) is private. 

@don41382 , the simplest workaroung might be to delay() proportional to the number of buffers (taking into account also the sampling rate and size of individual DMA buffer, etc...), before stop().

Or perhaps it will be more reliable if you fill the buffers (`out->consumeSample()`) with certain amount of zeroes, which should mean you 'pushed-out' the previous content . Again, number of zeros should be proportional to the number of DMA buffers and individual buffer length.

don41382, 2020-05-15:
@ikostoski I like your idea with filling the buffer with zeros. I will try this over the weekend.

Delay is a option too, but I would love to fix this for all esp32 users :-)

don41382, 2020-05-15:
Before I'll open up a PR, is this correct?

```
bool AudioOutputI2S::stop()
{
#ifdef ESP32
  int buffersize = this->hertz * this->dma_buf_count;
  for (int i=0;i<buffersize; i++) {
    int16_t samples[2] = {0x0,0x0};
    ConsumeSample(samples);
  }
  //i2s_zero_dma_buffer((i2s_port_t)portNo);
#endif
  return true;
}
```

ikostoski, 2020-05-15:
Hi @don41382,
It is actually easier to comment and suggest changes for the owner when you open the PR. 
My suggestions: 
- Add new method called `flush()` to be called before `stop()`. Sometimes stop means stop :).
- `buffersize` should be `64 * this->dma_buf_count`. It doesn't depend on the sample rate.
- Verify if you actually queued the sample by checking the return value of `ConsumeSample()`  and retry if it returns `FALSE` (i.e. sample was not queued), something like:
`while (!ConsumeSample(samples)) { delay(10); };`
- Move the `int16_t samples[2] = {0x0,0x0};` line above the loop.

Perhaps it should also be moved in the hierarchy above, i.e. in [AudioOutput.h](https://github.com/earlephilhower/ESP8266Audio/blob/master/src/AudioOutput.h), together with `buffersize`, but i think @earlephilhower should comment on that.

earlephilhower, 2020-05-15:
On the 8266, the DMA buffers are silenced out after they're used, anyway, so it was never a problem.  Your suggestion, @ikostoski , of breaking it into `::flush` and `::stop` makes a lot of sense.

don41382, 2020-05-16:
One last thing, while testing I figured out that the behaviour of a WAV file is different from a MP3 file generator. In `AudioGeneratorWAV.cpp` stop() is automatically called after the file has no more buffered data. After the discussion with @ikostoski, I think `stop()` should always be called explicitly. If you agree, I would open up another issue.

https://github.com/earlephilhower/ESP8266Audio/blob/e0d413622c829b06186684e3d61e8e2ffa0953f4/src/AudioGeneratorWAV.cpp#L98

don41382, 2020-05-26:
@ikostoski & @earlephilhower should this issue be closed since the PR is merged?

Thanks again for your support! And it would be nice if you give a quick feedback to my last question.

ikostoski, 2020-05-26:
Hi @don41382,

PR looked good so I did not comment further. I think that if you mention the issue (i.e. closes # xxx) within the PR, GitHub automatically closes the referenced issue when the PR is accepted. Otherwise @earlephilhower periodically cleans up.

Re: calling stop() explicitly, I would agree it should be called explicitly and perhaps more generators should checked if they do this at end of input. This is howerver @earlephilhower's project, so it is his opinion that counts, as he knows best how other generators work...

earlephilhower, 2020-05-30:
The issue here is the ESP32 is kind of broken IMO.  Instead of sending silence on underrun, it repeats the last buffers.  The ESP8266 just goes quiet when things are done.

It's not always correct to send (n) frames of 0 at the end of a sample.  Say, for example, you've got MP3 shuffle going on or playing prerecorded notes in multiple WAVs that you stitch together.  Adding n frames of 0s will make a relatively long silence between samples, in that case.

Right now, ESP32 apps would need to call flush() on their own to make that choice.  I think it's the only sane thing to do while leaving the flexibility for users.

@don41382, if you see some codecs where the `stop()` is missing, please do file a bug or PR.  It's supposed to be there, but I definitely could have missed them in some of the many deocders I did...

closed 2020-05-30 by earlephilhower