earlephilhower/ESP8266Audio #274: Easy play chunk

Voha888, 2020-06-06:
Hello.
Is it possible to play sound from the buffer, but without reference to any file source?
Those. so that I myself transfer a buffer in my scetch, for example 2048 bytes, and after playing it I transmit another?
So far, from the solutions I see only changing the source code of the library for my purposes, but maybe there is a ready-made solution?
thank

earlephilhower, 2020-06-06:
You can use the PROGMEM source, just pass in your RAM pointer and size.

closed 2020-06-06 by earlephilhower

Voha888, 2020-06-06:
Thank you very much for the help!

Voha888, 2020-06-07:
> You can use the PROGMEM source, just pass in your RAM pointer and size.

Hello! I encounter a strange problem. When I specify the entire track as a PROGMEM source, playback is correct.
But when I pass the pieces, then after some time the playback "hangs" on the line
`int ret = AACDecode (hAACDecoder, & inBuff, & bytesLeft, outSample);`
This is due to the fact that the variable
`int bytesLeft = buffValid;`
first it receives **1600 bytes**, then the value of the variable is constantly decreasing, and when the value becomes very small - the sketch and the sound hangs (hold).
If I add a check to this section of code, for example:
```
if (bytesleft <396) {
running = false;
return running;
}
```
That track will be played from start to finish, but a little incorrect, because it contains missed bytes.
My current test piece of code looks like this:

```

if (aac-> isRunning ()) {

    aac-> loop ();

  } else {

    in-> seek (0, SEEK_SET);

_// this cycle fills the RAM buffer with a new portion of bytes:_
    for (uint32_t i = 0; i <sizeof (mp3buff); i ++)
{
     mp3buff [i] = pgm_read_byte (& sampleaac [pos]);
     pos ++;
}
_// and the buffer is full again!_

    aac-> begin (in, out);
    
  
  }

```
where **mp3buff** is an array in RAM, **sampleaac** is a long track in PROGMEM
I want to make continuous playback of AAC \ MP3 from small pieces in RAM.

Please tell me, maybe I'm doing it wrong?

nicholmikey, 2022-01-31:
@Voha888 did you have success with this?

Voha888, 2022-01-31:
I don't remember well.  sorry