earlephilhower/ESP8266Audio #275: ESP32 from SD plays only first MP3 sample

Voha888, 2020-06-17:
Hello. I tried examples for playback from an SD card. And ESP8266Audio on my ESP32 only plays the first sample. If I use the buffer, it is empty, but not replenished from the SD card. Perhaps someone faced a similar problem? thank

ikostoski, 2020-06-17:
Can you please be more specific with the problem, i.e. post your example code and where you think it hangs ?

Voha888, 2020-06-17:
When I use playback from SPIFFS or PROGMEM, there are no problems. Here is my code for SDcard:

```
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceSD.h"
#include "AudioFileSourceID3.h"
static AC101 ac;  //need for AC101 out

AudioGeneratorMP3 *mp3;
AudioFileSourceSD *file;
AudioOutputI2S *out;
AudioFileSourceID3 *id3;

void play_from_SDcard(String filepath){

file = new AudioFileSourceSD(filepath.c_str());
 id3 = new AudioFileSourceID3(file);
 id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG"); //standart callback from examples
out = new AudioOutputI2S();
out -> SetPinout(IIS_SCLK /*bclkPin*/, IIS_LCLK /*wclkPin*/, IIS_DSIN /*doutPin*/);    //need for AC101 out
  mp3 = new AudioGeneratorMP3();
  mp3->begin(id3, out);

while(1){
if(mp3->isRunning()) {
    if (!mp3->loop())
    {
      mp3->stop();
      delay(2000);
      Serial.println("Stopped");
    }
  }
}
```

Serial Output:
```
+0 0x3ffb1d88
ID3 callback for: Performer = 'Iphone 5'
ID3 callback for: Title = 'IBest Ringtone'
Stopped
```
I hear the first second of the track, then Stopped. I tried using fileSourceBuffer:

```
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceSD.h"
#include "AudioFileSourceBuffer.h"
#include "AudioFileSourceID3.h"
static AC101 ac;

AudioGeneratorMP3 *mp3;
AudioFileSourceSD *file;
AudioFileSourceBuffer *buff;
AudioOutputI2S *out;
AudioFileSourceID3 *id3;

void play_from_SDcard(String filepath){

file = new AudioFileSourceSD(filepath.c_str());
buff = new AudioFileSourceBuffer(file, 40048);
    out = new AudioOutputI2S();
    
    out -> SetPinout(IIS_SCLK /*bclkPin*/, IIS_LCLK /*wclkPin*/, IIS_DSIN /*doutPin*/);  
  mp3 = new AudioGeneratorMP3();

mp3->RegisterStatusCB(StatusCallback, (void*)"mp3");
mp3->begin(buff, out);

while(1){

if(mp3->isRunning()) {
    if (!mp3->loop())
    {
      mp3->stop();
      delay(2000);
      Serial.println("Stopped");
    }else{
      buff->loop(); //this added in the hope that will help
    }
  }
delay(1);
}
```
The larger the buffer size I set, the longer the track plays. Serial Output:
```
+0 0x3ffb1d88
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 40048'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80096'
```
// - many similar lines - //
```
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80199'
STATUS(mp3) '260' = 'Decoding error 'reserved sample frequency value' at byte offset'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80201'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80202'
```
// - many similar lines - //
```
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80236'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80237'
STATUS(mp3) '260' = 'Decoding error 'reserved sample frequency value' at byte offset'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80239'
```
// - many similar lines - //
```
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80420'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80421'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80422'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80423'
STATUS(mp3) '257' = 'Decoding error 'lost synchronization' at byte offset 80424'
STATUS(mp3) '565' = 'Decoding error 'bad main_data_begin pointer' at byte offset 804'
STATUS(mp3) '565' = 'Decoding error 'bad main_data_begin pointer' at byte offset 808'
STATUS(mp3) '1' = 'Decoding error 'input buffer too small (or EOF)' at byte offset'
Stopped
```
Buffer after empty is not filled from SD card

ikostoski, 2020-06-17:
Can you please try to initialize `AudioOutputI2S` first (and set pinout), and after that initialize the SD card (`SD.begin`). If your SD card is connected to any of pins 26, 25 or 22, they are briefly taken over by I2S peripheral, and not returned to the SPI peripheral even if you move the I2S pins (with `SetPinout`)

How is the SD card connected ?

Voha888, 2020-06-17:
> How is the SD card connected ?

SD card pins: 22, 19, 23, 18, 5

AC101 pins (from Ai-thinker datasheet):  I2S_SDOUT **IO35**
                                                                 I2S_SDIN **IO25**
                                                                 I2S_LRCK **IO26**
                                                                 I2S_BCLK **IO27**
                                                                 I2S_MCLK **IO0**
                                                                          SCL **IO32**
                                                                         SDA **IO33**

> 26, 25 or 22,

Thank You, I will change and test

Voha888, 2020-06-17:
> pins 26, 25 or 22, they are briefly taken over by I2S peripheral, and not returned to the SPI peripheral even if you move the I2S pins (with `SetPinout`)

Thank you very much! It solved my problem

closed 2020-06-17 by Voha888