earlephilhower/ESP8266Audio #278: Support opus

SciLor, 2020-07-27:
It would be nice to be able to play opus files using libopus.

earlephilhower, 2020-07-30:
PRs are always welcome. :)

The opus codec looks like it can fit on the 8266 relatively simply, but because it is wrapped up in an OGG transport you end up porting opus, opusfile, and some portion of the ogg project as well.  Not impossible, just a lot of stuff unrelated to the actual codec.

closed 2020-08-12 by earlephilhower

GabeHC, 2021-02-22:
Opus sample code not working!
How do I fix it?

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Sample Opus playback begins...
+0 0x3ffb1ef8
abort() was called at PC 0x40081367 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40088340:0x3ffbe5c0 0x400885bd:0x3ffbe5e0 0x40081367:0x3ffbe600 0x4008138d:0x3ffbe620 0x400814e1:0x3ffbe650 0x400eb57b:0x3ffbe670 0x400e6f99:0x3ffbe930 0x400e6eac:0x3ffbe980 0x4008a5a5:0x3ffbe9b0 0x4008952d:0x3ffbe9e0 0x40082b04:0x3ffbea00 0x40083055:0x3ffbea30 0x400d689a:0x3ffae730 0x400d6a4f:0x3ffae7c0 0x400d7165:0x3ffae850 0x400d7bfb:0x3ffae8b0 0x400d995f:0x3ffafa20 0x400d5261:0x3ffb0d00 0x400d59f3:0x3ffb0e00 0x400d5f83:0x3ffb0ec0 0x400d60a7:0x3ffb1ea0 0x400e2109:0x3ffb1ed0 0x400e3ae5:0x3ffb1f00 0x400e3df1:0x3ffb1f50 0x400d3cde:0x3ffb1f70 0x400d1d45:0x3ffb1f90 0x400e5911:0x3ffb1fb0 0x40089596:0x3ffb1fd0

JonRobo, 2021-06-04:
Getting a similar error with the Opus example
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Sample Opus playback begins...
+0 0x3ffb1ef8
***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x40088aa0 on core 1

commit 0f5df48e (2020-08-12): Add Opus file format (and OGG demux)

Add ported versions of the Opus codec, OpusFile parser, and OGG demuxer
from Xiph.org.

Only works on the ESP32 due to memory limitations (of opusfile, not of
opus).

Adds a basic example and a host test.

Fixes #278
PR #294 (2020-08-12): Add Opus file format (and OGG demux)
Add ported versions of the Opus codec, OpusFile parser, and OGG demuxer
from Xiph.org.

Only works on the ESP32 due to memory limitations (of opusfile, not of
opus).

Adds a basic example and a host test.

Fixes #278
commit fae26d40 (2020-08-12): Add Opus file format (and OGG demux) (#294)

Add ported versions of the Opus codec, OpusFile parser, and OGG demuxer
from Xiph.org.

Only works on the ESP32 due to memory limitations (of opusfile,
strangely enough, and not because of opus compression).

Adds a basic example and a host test.

Fixes #278