earlephilhower/ESP8266Audio #279: 16MB ESP32 SPIFF WAV playback error (fine on 4MB)

zachtos, 2020-07-28:
I'm working with 16MB version of ESP32 and notice audio distortion when using larger SPIFF partitions.  There is a long artifact/distortion near the end of any file I play back.  It goes away if I switch back to a 4MB version of dev kit.  I am out of ideas why it would work differently on a larger partition?
#Name,	Type,	SubType,	Offset,	Size,	Flags
nvs,	data,	nvs,	0x9000,	0x5000,	
otadata,	data,	ota,	0xe000,	0x2000,	
app0,	app,	ota_0,	0x10000,	0x300000,	
spiffs,	data,	spiffs,	0x310000,	0xCF0000,  //does not play back properly when 0xCF0000, but 0xF0000 is fine for spiff size when playback audio.

`#include <Arduino.h>
#include <WiFi.h>
#include "SPIFFS.h"
#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2S.h"
#include <HTTPClient.h>
#include <SD.h>

AudioGeneratorWAV *wav;
AudioFileSourceSPIFFS *file;
AudioOutputI2S *out;

unsigned long recoil;

void setup() {
  // put your setup code here, to run once:
WiFi.mode(WIFI_OFF);
  Serial.begin(115200);
  delay(1000);

  //audio file system
  SPIFFS.begin();
  Serial.printf("Sample Wav playback begins...\n");

  file=new AudioFileSourceSPIFFS("/E08.wav");
  out=new AudioOutputI2S(0,0,32);//0, 0, 22);//16); //0, 0, 64); 
  //out->SetPinout(26, 25, 22); //int bclkPin, int wclkPin, int doutPin) desired to keep IR pins on old PCB in tact
  wav=new AudioGeneratorWAV();
  wav->begin(file, out);
}

void loop() {
  // put your main code here, to run repeatedly:
 
  if (wav->isRunning()) //will crash if do audio while sending I think
    {
    if (!wav->loop()) 
    {
      wav->stop();
    }
  }

}`

earlephilhower, 2020-07-29:
Dupe of #136

closed 2020-07-29 by earlephilhower

zachtos, 2020-07-29:
Update:  LittleFS did make it better, but now files have a slower opening speed.  They read fast once open, which removes the glitches, but I was opening files rapidly (like once per 75ms for 200kB WAV).  I wonder if there is a way to improve the opening speed?

  out=new AudioOutputI2S(0,0,12);//16 ok on little fs, higher gives smoother but slows down, too short = glitch
  mixer=new AudioOutputMixer(24, out); //64 buffer seems ok no issues, higher = smoother but slows clock