earlephilhower/ESP8266Audio #282: ESP32 Webradio [WiFiClient.cpp:482] flush(): errno: 11, "No more processes"

klaus-liebler, 2020-08-04:
**Environment**
- Windows 10, VS Code, PlatformIO, 
- Chinese ESP32 DevBoard
- 1-Transistor-Amplifier connected on GPIO25 (Hardware works; tested with slightly modified "PlayWavFromProgmen" example)

**Error Log**

Connecting to WiFi
...Connecting to WiFi
Connected
Go to http://192.168.1.207/ to control the web radio.
+0 0x3ffb1f28
Running for 2 seconds ...Free mem=161260
Running for 3 seconds ...Free mem=161260
Running for 4 seconds ...Free mem=161260
Running for 5 seconds ...Free mem=161260
Running for 6 seconds ...Free mem=161260
Running for 7 seconds ...Free mem=161260
Running for 8 seconds ...Free mem=161260
Running for 9 seconds ...Free mem=161260
[E][WiFiClient.cpp:482] flush(): fail on fd 55, errno: 11, "No more processes"
Sending INDEX...Free mem=160120
Sent INDEX...Free mem=160480
[E][WiFiClient.cpp:482] flush(): fail on fd 56, errno: 11, "No more processes"
Sending INDEX...Free mem=159624
Sent INDEX...Free mem=159216
[E][WiFiClient.cpp:482] flush(): fail on fd 57, errno: 11, "No more processes"
Sending INDEX...Free mem=159748
Sent INDEX...Free mem=157204
[E][WiFiClient.cpp:482] flush(): fail on fd 58, errno: 11, "No more processes"
Sending INDEX...Free mem=159304
Sent INDEX...Free mem=159796
Running for 10 seconds ...Free mem=159796
Running for 11 seconds ...Free mem=159796
Running for 12 seconds ...Free mem=159796

**Steps to reproduce:**
- Get "WebRadio"-Example
- Configure WIFI SSID and Password
- Changed line 296 to `out = new AudioOutputI2S(0, AudioOutputI2S::INTERNAL_PDM);`  (to fit my hardware ...btw, the same error occurs without modification)
- Compile, Upload, Monitor, Open Website

earlephilhower, 2020-08-04:
Sorry, sounds like an issue in the ESP32 WiFiClient().  That part's just used, not modified in the examples.  You might get support in the ESP32 Arduino project...

klaus-liebler, 2020-08-05:
I investigated a little bit more in this issue. I commented out line 181 in web.cpp (client->flush()) and now at least it works :-) I am not aware about the long term consequences. But  as I just need a technology demostration, I am happy with this solution...

closed 2020-08-05 by klaus-liebler