earlephilhower/ESP8266Audio #286: How to use mp3 and aac at the same time?

Voha888, 2020-08-06:
Hello!
I want to initialize 2 decoders - mp3 and aac.
Right now I only use mp3:

```
out = new AudioOutputI2S ();
    
out -> SetPinout (IIS_SCLK / * bclkPin * /, IIS_LCLK / * wclkPin * /, IIS_DSIN / * doutPin * /);

in = new AudioFileSourceCHUNKPLAYER (mp3buff, 2050); _// this is AudioFileSourcePROGMEM, but the read method is slightly changed here - it reads the chunks of audio, which I need from the necessary sources (sd, spiffs, radio)_

 mp3 = new AudioGeneratorMP3 ();

 mp3-> begin (in, out);
```

Everything works very well now, _(thanks to earlephilhower for the great advice https://github.com/earlephilhower/ESP8266Audio/issues/274)_

But when I add **AAC** to read it piece by piece, my ESP starts to reboot cyclically.
 I want to write like this:

```
out = new AudioOutputI2S ();
    
out -> SetPinout (IIS_SCLK / * bclkPin * /, IIS_LCLK / * wclkPin * /, IIS_DSIN / * doutPin * /);

in = new AudioFileSourceCHUNKPLAYER (mp3buff, 2050); _// this is AudioFileSourcePROGMEM, but the read method is slightly changed here - it reads the pieces of audio I need from the necessary sources (sd, spiffs, radio)_

 mp3 = new AudioGeneratorMP3 ();

 mp3-> begin (in, out);

aac = new AudioGeneratorAAC ();

aac-> begin (in, out);
```

Then to put my piece of audio into mp3buff, and `call mp3-> loop ();` or `aac-> loop ();` depending on the chunk format.
Can you do this?

earlephilhower, 2020-08-08:
No, that's not generally possible.  When starting a new AAC stream, make the AAC decoder and hook things up.  Then when done and an MP3 comes is, make that and run it.  They both can't run because of things like output format (mono/stereo/rate) are not guaranteed the same and because both MP3 and AAC use a LOT of memory and you could end up OOM very easily on even the ESP32.

closed 2020-08-08 by earlephilhower

Voha888, 2020-08-09:
> because both MP3 and AAC use a LOT of memory and you could end up OOM very easily on even the ESP32.

I have replaced the `malloc `calls with `ps_mallo`c to use the extra psram memory on the **ESP32-A1S** module. I think such an option can be provided in the library, and use psram as an optional parameter when initializing an instance of a class or starting playback. Can I create a pull request if you like?