earlephilhower/ESP8266Audio #301: esp32 flac slower sound

CrocKlok, 2020-09-02:
Hello @earlephilhower I have the mp3 and the wave perfectly working from sd card. I have question about how to resolve flac sound that sound slower. I try multiple flac files also from https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/audio-samples.html . It also read from a sd card I also play with a SPI, speeds and nothing helped for me. Was wondering how to resolve it. 

I try example the playflafromprogmemtodac and works without any problems. I was wondering how to make that kind of .h files in windows 7 ?

earlephilhower, 2020-09-03:
To make the H file you need something like the "xxd" utility.  It may be available on the Gnu32 Windows utilities, or you can use the WSL (Windows Subsystem for Linux) to generate them.  

For FLAC, are you getting a stuttering playback , or it's playing at, say, 1/2 rate and 2x lower in frequency?  If it's stuttering, that's due to the IO speeds and you may need to reduce the BW of the FLAC file until the ESP32 flash can keep up.  If it's outputting things at the wrong frequency, can you find a small sample ant attach it here?  There are host tests which use the exact same code and can be used to see if there's something in the FLAC or if there's something in the ESP32 I2S frequency.

CrocKlok, 2020-09-03:
Here an example it is this file I play from this website.

https://dl.espressif.com/dl/audio/gs-16b-2c-44100hz.flac

[flac test.zip](https://github.com/earlephilhower/ESP8266Audio/files/5169328/flac.test.zip)

earlephilhower, 2020-09-03:
Sorry, the FLAC is decoding fine from the recording, the data transfer is just too slow.  You can hear the stutters in your recording, and the host test I just added gives the correct output.  You need to use a faster filesystem or flash.  I don't do much ESP32 work, so can't really suggest the best way to speed up the I/O.  On the 8266 we would use LittleFS which is 8x faster than SPIFFS.  Good luck!

earlephilhower, 2020-09-03:
Or, it could be that you're not calling the `flac->loop()` fast enough.  If you're not doing just the example (i.e. you're running a full sketch and doing things like LCD refreshes or web stuff between calls to `loop()`) then sprinkle in more `flac->loop()` calls in the app.  If there's no work to be done, those calls won't do anything and use ~0 CPU.

closed 2020-09-03 by earlephilhower

CrocKlok, 2020-09-03:
I do just the example and change it to read from the ds card. I don't have any lcd or web stuff. This is pure read from the sd card. is there a way use littleFS on the esp32 ?

CrocKlok, 2020-09-04:
Hmm update. I got the xxd working and store it as a progmem and works pretty nicely and the sound was also good. the only problem was had to crop the music short time because of sketch get to big. Now I am still trying to get an sd card in a good way working. I had some improvement after changing some option but I still have some chopper sound in the back. I almost think that the sd lib is buggy. I use a different way to get the sd card reading to the readout. Maybe it has to do with that?

I saw in the example you use:
SD.begin(SS, SPI_SPEED);

I get some problems because it doesn't know SPI_SPEED SD_SCK_MHZ(40) all kinds of errors even it is declared at the top.

I use this way: SD.begin(33,SPI,40000000)

that works for me but I might think the sd lib is buggy. Is this possible? I can use some help here if possible @earlephilhower

earlephilhower, 2020-09-04:
Sorry, I'm really not up on the ESP32's I/O stuff.  On the 8266 I ported a newer version of the SD library and the LittleFS because of the SPIFFS/old SD.h bad performance.

For SD, I'd first try a different card as that's easiest.  Then, see if you can use HW SPI to do the SD access...it's possible to bit-bang it, but that eats CPU cycles that are needed for playback.  

LittleFS, you can try https://github.com/lorol/LITTLEFS .  I haven't personally used it, but it looks like it might just be the thing.

Good luck!

CrocKlok, 2020-09-06:
Thank you @earlephilhower I would try it out. I will also try first the esp8266 if that makes more sense. So I could test this out and get the flac working. I will keep it posted if I know more.