earlephilhower/ESP8266Audio #316: Has anybody tried this with an LM386?

fgolemo, 2020-10-19:
Hey folks,

First off, huge thanks for this library. Really great work!

I'm trying to run this with an old-school LM386 module a la https://www.amazon.ca/Oiyagai-5V-12V-Amplifier-Module-Arduino/dp/B07BXPWKSR/ and I'm getting mostly noise with a hint of the audio I'm playing back. 

I've wired the Rx of my ESP8266 (NodeMCU) to the speaker positive, ground to ground, 5V input from an external power supply, and connected the ground of the NodeMCU with the ground of the PSU. No extra resistors/capacitors or anything. 
Using AudioOutputI2SNoDAC. 

Do I need a dedicated I2S amp and can't use a simple signal amp?

Cheers,
Florian

earlephilhower, 2020-10-19:
That amp is not going to work w/the built-in hardware, unfortunately.  If you have an I2S DAC, you can use the I2SOutput class and take the DAC's small signal output as input to the LM386 and it'll amplify, just like a normal signal.

Using the I2SNoDAC, you can't use the LM386 (or any analog amp) and get good results because the 8266 isn't sending an analog voltage.  You'd need the 1T digital amplifier setup as discussed in the README.  OTW the 386 will end up seeing a pulse train (at up to 1MHZ!) of 0-3.3V and go into an unhappy state.

closed 2020-10-19 by earlephilhower

fgolemo, 2020-10-19:
Ah gotcha, thanks. 👍

earlephilhower, 2020-10-19:
Actually, there's a *but* about the LM386.  If you pass the ESP8266 digital pulse stream through an external LPF (RC, etc.) and then pass that to the LM386 you might be able to get something that's passable.  You'll have to build an appropriate LPF, of course, and simple RC networks might let through enough high frequency artifacts to make it unusable.  I haven't personally tried it, but, if you've already got the amps and some discretes then it can't hurt to give it a try.