earlephilhower/ESP8266Audio #322: How to play two separate mono audio files simultaneously on DAC1 and DAC2 on ESP32?

ThePiGuy, 2020-10-27:
I want to play two separate mono audio files simultaneously, with one playing on DAC1 and the other playing on DAC2. I've considering trying something with a mixer, but am not totally sure where to start. Any advice? 

What if I only wanted to play one mono file at a time to either DAC1 or DAC2? Any suggestions on how best to specify which DAC channel to play to? Thanks!

earlephilhower, 2020-10-27:
Just make two separate I2S objects, one for DAC1, one for DAC2, and connect them to two separate decoders (WAV/etc.  MP3/AAC may have CPU limitations) and everything should work.  You'll call `obj1->loop(); obj2->loop();` in your main loop, of course.

closed 2020-10-27 by earlephilhower

ThePiGuy, 2020-10-27:
This is the line given in the examples to create an I2S object for the ESP32's onboard DAC: `out = new AudioOutputI2S(0, 1);`. This appears to create a mono audio output device that outputs to both GPIO25 (DAC1) and GPIO26 (DAC2). Is there a way to create this `AudioOutputI2S` object for only DAC1/DAC2 separately?

earlephilhower, 2020-10-27:
Ah, didn't catch that, thought you wre using internal DAC and external DAC.

Looking at the ESP32 docs, I don't *think* you can DMA separately to either one.  You'll need to investigate the IDF and see if there's a way to set that up.  If so, you'll probably need to adjust the ESP32 I2S output object to do the magic.