earlephilhower/ESP8266Audio #325: Sound driver for ESP32 not using I2S (uses ULP processor)

Martin-Laclaustra, 2020-11-01:
Thank you for all your shared work!
I adapted @bitluni 's code in https://github.com/bitluni/ULPSoundESP32 to work as sound output driver for this library.
It allows using both DACs (GPIO25 and GPIO26) in stereo mode or select only one of them in mono mode.
I2S remains free for other uses.
Relevant usage notes:

```
#include "AudioOutputULP.h"
AudioOutputULP *out = NULL;
void setup()
{
  out = new AudioOutputULP(); // stereo
  //out = new AudioOutputULP(1); // mono, only DAC 1
  //out = new AudioOutputULP(2); // mono, only DAC 2
  ...
}
```
[AudioOutputULP.h.txt](https://github.com/earlephilhower/ESP8266Audio/files/5470346/AudioOutputULP.h.txt)
[AudioOutputULP.cpp.txt](https://github.com/earlephilhower/ESP8266Audio/files/5470347/AudioOutputULP.cpp.txt)

Let me know if you are interested in adding these to the library.

earlephilhower, 2020-11-01:
That's pretty neat, it's be a good addition!

Would you like to make a real pull request, or would you like me to manually add it from your attachment?

Martin-Laclaustra, 2020-11-01:
I am ok with you adding it directly.
In the future, if I work on more complex contributions to the library I would go the whole "regular" path. For these kind of drivers I think this is a "lighter" way of doing it.

earlephilhower, 2020-11-01:
I had a couple spots where I needed int->uint32_t conversions to make GCC -Wall happy, but I don't expect it had any actual code effect.  If you could give it a quick runthrough to verify it's still good, that would be helpful.  Thx for the contribution!

closed 2020-11-01 by earlephilhower

Martin-Laclaustra, 2020-11-01:
The changes look good. I will test it tomorrow.
Reviewing them I discovered that current code swaps the order of the samples in mono mode.
```
      unsigned int w = ms[0];
      if(stereoOutput){
        w |= ms[1] << 8;
      } else {
        w |= bufferedOddSample << 8;
        bufferedOddSample = 128;
        waitingOddSample = false;
      }
```
This is the fix: (but I will test this code tomorrow too)
```
      unsigned int w = ms[0];
      if(stereoOutput){
        w |= ms[1] << 8;
      } else {
        w = bufferedOddSample;
        w |= ms[0] << 8;
        bufferedOddSample = 128;
        waitingOddSample = false;
      }
```

Martin-Laclaustra, 2020-11-02:
I confirm that your changes and my proposed fix work as they should.
The only missing thing is adding 3 lines of usage (like in the OP) to the top of the header file. Otherwise, I am afraid that the mono feature will remain hidden/unknown for most users.

Thank you for the warmth attention and the quick adoption.

roboticboyer, 2020-11-14:
@earlephilhower please could you submit a new release of the library with ULP feature and an example of the webradio
Thank you
Gio

earlephilhower, 2020-11-14:
@roboticboyer , you should be all set!