earlephilhower/ESP8266Audio #327: [possible bug] PlayRTTTLToI2SDAC example in ESP32 does not clear last note

Martin-Laclaustra, 2020-11-02:
The last note keeps playing forever. The only change to the example was to enable internal DAC.
`out = new AudioOutputI2S(0,1);`
This does not happen in ESP8266.

earlephilhower, 2020-11-14:
Interesting observation.  I see that the code does call `out->stop()` which should fill the I2S buffer with 0s and also stop the I2S engine.  Unfortunately, I'm not really knowledgeable about the internal DAC.  If there is something else required to stop samples going to it, that may be missing.

fkoteam, 2020-12-12:
after stop a wav, I start to hear noise. It can be this bug? any workaround?
(with esp32 and internal dac, also)

thank you!

schmurtzm, 2021-12-25:
Hi, I can confirm that the problem concerns the ESP32 only (tested on ESP8266,  Wemos esp32 and devkit v4) : I take the example "PlayRTTTLToI2SDAC.ino" included in the library, I flash my ESP32 and it works but the last note is repeated for ever.

It's not due to the internal DAC because the same problem appear with no DAC or with an external DAC.

Some observations from my issue #471 which is the same problem (sorry the search engine of github doesn't find this issue if you type "RTTTL", you have to type "PlayRTTTLToI2SDAC" 😄 ) :

- The ESP32 is not crashed.
- If you call rtttl->stop(); before the end of the RTTTL sound then it stops correctly.
- If you call rtttl->stop(); after the end of the RTTTL sound then it doesn't stop the repeat of the last note.

I've recorded [the result sound from my ESP32 here](https://dl.dropbox.com/s/i4y3qrln1kw2exw/RTTTL_on_ESP32.mp3).

Hope this helps 😉

commit fe7a1f79 (2021-12-27): Fix RTTTL stop on ESP32 (#473)

Fixes #327 
Fixes #471
closed 2021-12-27 by earlephilhower