earlephilhower/ESP8266Audio #329: How to play multiple wav files without interruption?

sehraf, 2020-11-07:
Hi there,

i'm currently trying to implements a [Rhasspy](https://rhasspy.readthedocs.io/en/latest/) satellite using the MQTT based [Hermes](https://rhasspy.readthedocs.io/en/latest/audio-output/#mqtthermes) protocol on an ESP32 Audio Kit. The protocol sends 2k chunks of WAV data over MQTT. Each chunks is a **complete WAV file** with header.

My first approach was to use an AudioFileSourcePROGMEM -> AudioGeneratorWAV -> AudioOutputI2S piepline, which worked except it stutters on each new chunk. For each new WAV file/chunk i'm calling `AudioFileSourcePROGMEM::open()` on the data stored in RAM and then `AudioGeneratorWAV::begin()`.
After a few days fiddling around i found the problem: `stop()`!

I have  stutter free playback when adding an AudioOutput layer between AudioGeneratorWAV and AudioOutputI2S  that drops calls to `stop()`.

So long story short, is there a "better" way to play many small WAV files without interruption? Or is the extra layer the way to go?  Not calling `AudioGeneratorWAV::stop()` will also not call `AudioOutputI2S::stop()` but leak its internal buffer to my understanding.

I've also added an AudioOutputBuffer to compensate WiFi/MQTT hiccups, works really nice now :rocket: Thanks for the handy library!

kostuch, 2020-12-04:
`I have stutter free playback when adding an AudioOutput layer between AudioGeneratorWAV and AudioOutputI2S that drops calls to stop().`

Could you please share your modifications, please?
I have very similar problem with small wavs transmitted by espnow.
Even if I try to re-play already defined wav from RAM or FLASH in loop (without any additional network setup), there is distortion on each pass.

Regards

sehraf, 2020-12-04:
sure
https://gist.github.com/sehraf/75aa1cfe2609367f71418bf2eacf7da7
https://gist.github.com/sehraf/0cccd780302bd7386c9c0d9f02c4097b

Put it between the generator and the output.
This also catches all calls to setRate and similar.