earlephilhower/ESP8266Audio #330: AudioOutputI2S constructor calls SetPinout on conflicting pins

Adamcbrz, 2020-11-12:
I have multiple sensor running on the same esp32 board. I was utilizing an IR sensor on pin 25 before integrating sound. Since I knew my pin 25 was taken I testing this library on pins (14,27,26). Everything worked in this configuration when testing by itself. I then ran into issues when integrating my IR sensor with this library. The problem is that even though I was calling SetPinout(14,27,26) it was taking over my 25 pin because the AudioOutputI2S constructor was calling SetPinout on pin (26, 25, 22).

My current solution was to modify the source file to use the pins I needed. I don't like this solution for the longer term because it will break the next time the library is updated. 

I don't mind doing a pull request but wanted to know what would be the preferred method of implementing a possible fix for this.

earlephilhower, 2020-11-12:
I see your problem.  Kind of a catch-22 situation.

Looking thru the code I think the I2S setup needs to move out of the constructor and into the `::begin` method.  That's probably the correct way to do it, in any case.  Sorry, this library kind of grew and has a few warts like that.

A PR would be appreciated if you'd like to make it lazy-initialize in ::begin.  Things like ::SetPins/etc. would then just set class variables which would be used in the `::begin` method to actually start the I2S.

Adamcbrz, 2020-11-12:
> I see your problem. Kind of a catch-22 situation.
>
> Looking thru the code I think the I2S setup needs to move out of the constructor and into the `::begin` method. That's probably the correct way to [...]

I will be glad to do the PR but wanted to clarify before moving forward.

You are suggesting storing the pin numbers as class variables and calling SetPinout in the `::begin` method. Using either the default pins or what was previously submitted using SetPinout. Is that correct?

As an alternate to storing additional values. What if we modify the constructor to include those parameters with default values. 
`AudioOutputI2S(int port=0, int output_mode=EXTERNAL_I2S, int dma_buf_count = 8, int use_apll=APLL_DISABLE, int bclk=26, int wclk=25, int dout=22);`

Either method works so which one would you prefer?

earlephilhower, 2020-11-12:
> You are suggesting storing the pin numbers as class variables and calling SetPinout in the ::begin method. Using either the default pins or what was previously submitted using SetPinout. Is that correct?

A little bigger than that, actually.  The constructor should only set class vars, and any actions would happen in the ::begin (i.e. i2s_begin, setting the mode, pins, etc.).  That would be more in keeping with the Arduino begin/end paradigm and cleaner.

It is a bit of work (I don't think much, though...mostly just move guts from the constructor to begin) so if that's a problem no worries and now that you've planted that seed I'll eventually get around to fixing it myself. :)

Adamcbrz, 2020-11-12:
No worries, I will try to get to it tomorrow.

commit 299db3d0 (2020-11-14): Modify AudioOutputI2S to initialize i2s during begin (#331)

Fixes #330
closed 2020-11-14 by earlephilhower