earlephilhower/ESP8266Audio #339: STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value''

k4r4c, 2020-11-20:
Hi!

I had some very exciting moment playing around with this library. I got the streaming hooked up with a pcm5102, and got a pretty nice result. However, it stutter sometimes, so I decided to extend the ram with a 23lc1024 chip. I have a nodemcu with esp-12E.
I have the following issue:

```
Connected
SPI RAM buffer size: 131072 Bytes
Running for 7561 ms...
STATUS(buffer) '999' = 'Filling buffer...'
METADATA(ICY) 'StreamTitle' = 'Troy Trombone Shorty Andrews - It Ain't No Use'
STATUS(buffer) '999' = 'Buffer filled...'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 0'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 9821'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 19338'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 28775'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 38228'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 47597'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 56971'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 66336'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 75671'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 85022'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 94319'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 103573'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 112660'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 121736'
STATUS(buffer) '999' = 'Filling buffer...'
STATUS(buffer) '999' = 'Buffer filled...'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 129538'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 138784'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 147851'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 156896'
STATUS(mp3) '259' = 'Decoding error 'forbidden bitrate value' at byte offset 165924'
```
it might be related to [this issue](https://github.com/earlephilhower/ESP8266Audio/issues/72), but in my case, the [test](https://github.com/Gianbacchio/ESP8266_Spiram/blob/master/examples/ESP_spiram/ESP_spiram.ino) with the spi ram alone works.
I compiled and flashed StreamMP3FromHTTP_SPIRAM slightly modified with the dac connected (also I made sure that cs is connected to GPIO0 which is D3 on the nodemcu):
```
#include <Arduino.h>
#ifdef ESP32
    #include <WiFi.h>
#else
    #include <ESP8266WiFi.h>
#endif
#include "AudioFileSourceICYStream.h"
#include "AudioFileSourceSPIRAMBuffer.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

// To run, set your ESP8266 build to 160MHz, update the SSID info, and upload.

// Enter your WiFi setup here:
#ifndef STASSID
#define STASSID "xxxx"
#define STAPSK  "xxxx"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

// Randomly picked URL
const char *URL="http://jazz.streamr.ru/jazz-64.mp3";

AudioGeneratorMP3 *mp3;
AudioFileSourceICYStream *file;
AudioFileSourceSPIRAMBuffer *buff;
AudioOutputI2S *out;

// Called when a metadata event occurs (i.e. an ID3 tag, an ICY block, etc.
void MDCallback(void *cbData, const char *type, bool isUnicode, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);
  (void) isUnicode; // Punt this ball for now
  // Note that the type and string may be in PROGMEM, so copy them to RAM for printf
  Serial.printf_P(PSTR("METADATA(%s) '%s' = '%s'\n"), ptr, type, string);
  Serial.flush();
}

// Called when there's a warning or error (like a buffer underflow or decode hiccup)
void StatusCallback(void *cbData, int code, const char *string)
{
  const char *ptr = reinterpret_cast<const char *>(cbData);
  static uint32_t lastTime = 0;
  static int lastCode = -99999;
  uint32_t now = millis();
  if ((lastCode != code) || (now - lastTime > 1000)) {
    Serial.printf_P(PSTR("STATUS(%s) '%d' = '%s'\n"), ptr, code, string);
    Serial.flush();
    lastTime = now;
    lastCode = code;
  }
}

void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.println("Connecting to WiFi");

  WiFi.disconnect();
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
  
  WiFi.begin(ssid, password);

  // Try forever
  while (WiFi.status() != WL_CONNECTED) {
    Serial.println("...Connecting to WiFi");
    delay(1000);
  }
  Serial.println("Connected");

  audioLogger = &Serial;
  file = new AudioFileSourceICYStream(URL);
  file->RegisterMetadataCB(MDCallback, (void*)"ICY");
  buff = new AudioFileSourceSPIRAMBuffer(file, 0, 131072);
  buff->RegisterStatusCB(StatusCallback, (void*)"buffer");
  out = new AudioOutputI2S();
  mp3 = new AudioGeneratorMP3();
  mp3->RegisterStatusCB(StatusCallback, (void*)"mp3");
  mp3->begin(buff, out);
}

void loop()
{
  static int lastms = 0;

  if (mp3->isRunning()) {
    if (millis()-lastms > 1000) {
      lastms = millis();
      Serial.printf("Running for %d ms...\n", lastms);
      Serial.flush();
     }
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}
```

shalnoff, 2022-02-23:
Same issue. Separate RAM test works just fine. CS changing (including original HCS) doesn't help. 
@k4r4c did you find any solution?

thekeels, 2023-06-30:
Same issue for me. @k4r4c @shalnoff