earlephilhower/ESP8266Audio #341: Play samples from RAM buffer

kostuch, 2020-11-30:
Hi,
I wonder if this library can be used to play raw samples from RAM memory.
I have a project to transmit real time sound samples using espnow and it looks promising.
So far I created additional classes based on existing progmem and wav, but I have lot of artifacts and don't know the reason.
AudioFileSourceRAM - to simulate "file" in RAM buffer filled by espnow receive callback
AudioGeneratorRAW - to play raw samples (8kHz samplerate, mono, 8bit)

Can you advise if this is proper approach or I should use somehow existing possibilities.
I can provide my files and sketch for review

Thank you

earlephilhower, 2020-11-30:
FYI, there's no need for AudioFileSourceRAW, really.  You can use AudioFileSourcePROGMEM with any buffer, including RAM if desired.

Without seeing your code this is only a guess, but make sure you are promoting 8-bit to 16-bit properly.  Some 8 bit formats are unsigned 0...256, others are -128..127.  16-bit is always signed -32768...32727.  The WAV decoder has this conversion in the code, so you can look there for an example.

Otherwise, make sure you're calling the play->loop() routine often enough so that the I2S buffer doesn't underflow.  If you don't you'll get clicks or tapping in the output since missing I2S data will be filled w/0s.

Good luck!

earlephilhower, 2020-11-30:
Actually, re-reading your question, if you can add a WAV header (it's simple to make) to the buffer, using the WAVGenerator might be your best, least-work option.  The WAV format is very simple and once you have the template it's just a matter of specifying the length properly to re-use it on other samples.

closed 2020-11-30 by earlephilhower

kostuch, 2020-11-30:
Hi,
Thank you for the prompt answer.
Regarding formats, I just make ADC (from microphone or other audio source) on the espnow sender side at 8kHz and send 250 8bit samples to receiver (250 is max packet size). As the phoneline quality is enough for me, this is sufficient.
Then trying to DAC these samples on the espnow receiver in realtime.
I already tried to use native sigmadelta feature of ESP8266/ESP32 with sort of success but your lib is much more flexible.
That's why I need some hints how to hookup espnow receive buffer to audio sink.
The AudioFileSourcePROGMEM class has method 'read' implemented using memcpy_P, so I doubt if I can use it with RAM buffer without overloading?

I attached sketch and (dirty hacked) classes for RAM/RAW audio source.
With such simple program I noticed that first buffer is played cleanly and other ones with clicking sound (caused by stop/start audio generator in loop() maybe???). So my classes somehow works, but I misuse them.

[test_audio.zip](https://github.com/earlephilhower/ESP8266Audio/files/5618598/test_audio.zip)

earlephilhower, 2020-12-01:
>  The AudioFileSourcePROGMEM class has method 'read' implemented using memcpy_P, so I doubt if I can use it with RAM buffer without overloading?

No, it will work fine on RAM or ROM.  `memcpy_P` is safe no matter what source address location.

I don't know exactly what espnow is, but it could be that it is taking up large chunks of CPU time.  Your general outline seems fine, but the time from one `loop()` call to the next is too big.  As a sanity check, I'd drop the espnow for a test and just use a simple generated buffer (say, a sine wave or something) and see how things sound.

kostuch, 2020-12-02:
> I don't know exactly what espnow is, but it could be that it is taking up large chunks of CPU time. Your general outline seems fine, but the time from one `loop()` call to the next is too big. As a [...]

I removed all espnow related parts and try to play RAM buffer filled with raw samples (just sine wave).
Effect is exactly the same - clicking distortion after each buffer pass.
Ideally RAM buffer should be played over and over without any pause in between. So, it's something related to stop/open/begin

This is my simple test sketch:
```
#include <Arduino.h>
#include "ESP8266HTTPClient.h"
#include "SD.h"
#include "AudioFileSourceRAM.h"
#include "AudioGeneratorRAW.h"
#include "AudioOutputI2SNoDAC.h"

AudioFileSourceRAM *file;
AudioGeneratorRAW *raw;
AudioOutputI2SNoDAC *out;

#define MSG_SIZE 250
uint8_t rcv_buff[MSG_SIZE] = // sinus 440Hz wave samples @8kHz sample rate, should be played every 32ms
{
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84, 
	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18
};

void setup()
{
	Serial.begin(115200);
	delay(1000);
	audioLogger = &Serial;
	file = new AudioFileSourceRAM(rcv_buff, MSG_SIZE);
	out = new AudioOutputI2SNoDAC();
	raw = new AudioGeneratorRAW();
	raw->begin(file, out);
}

void loop()
{
	if (raw->isRunning()) raw->loop();
	else
	{
		file->open(rcv_buff, MSG_SIZE);   // re-open samples buffer
		raw->begin(file, out);   // play again
	}
}

```

Hellforce, 2021-12-11:
> > Esnow'un tam olarak ne olduğunu bilmiyorum, ancak büyük miktarda CPU zamanını alıyor olabilir. Genel taslağınız iyi görünüyor, ancak bir `loop()`aramadan diğerine geçen süre çok uzun. Akıl sağlığı kontrolü olarak, bir test için espnow'u bırakırdım ve sadece basit bir oluşturulmuş arabellek kullanırdım (örneğin, bir sinüs dalgası veya başka bir şey) ve işlerin nasıl ses çıkardığını görürdüm.
> 
> Esnow ile ilgili tüm parçaları kaldırdım ve ham örneklerle dolu RAM tamponunu oynatmaya çalıştım (sadece sinüs dalgası). Etki tamamen aynıdır - her arabellek geçişinden sonra tıklama bozulması. İdeal olarak RAM arabelleği, aralarında herhangi bir duraklama olmaksızın tekrar tekrar oynatılmalıdır. Yani, dur/aç/başla ile ilgili bir şey
> 
> Bu benim basit test çizimim:
> 
> ```
> #include <Arduino.h>
> #include "ESP8266HTTPClient.h"
> #include "SD.h"
> #include "AudioFileSourceRAM.h"
> #include "AudioGeneratorRAW.h"
> #include "AudioOutputI2SNoDAC.h"
> 
> AudioFileSourceRAM *file;
> AudioGeneratorRAW *raw;
> AudioOutputI2SNoDAC *out;
> 
> #define MSG_SIZE 250
> uint8_t rcv_buff[MSG_SIZE] = // sinus 440Hz wave samples @8kHz sample rate, should be played every 32ms
> {
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84,
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18, 46, 84, 
> 	128, 171, 209, 237, 253, 253, 237, 209, 171, 128, 84, 46, 18, 2, 2, 18
> };
> 
> void setup()
> {
> 	Serial.begin(115200);
> 	delay(1000);
> 	audioLogger = &Serial;
> 	file = new AudioFileSourceRAM(rcv_buff, MSG_SIZE);
> 	out = new AudioOutputI2SNoDAC();
> 	raw = new AudioGeneratorRAW();
> 	raw->begin(file, out);
> }
> 
> void loop()
> {
> 	if (raw->isRunning()) raw->loop();
> 	else
> 	{
> 		file->open(rcv_buff, MSG_SIZE);   // re-open samples buffer
> 		raw->begin(file, out);   // play again
> 	}
> }
> ```

Sorry for writing after a long time. I am developing a project using ESPNOW DAC. Were you able to find a solution to the problem you are experiencing? Can you share your sample codes with me kostuch

kostuch, 2021-12-11:
Sorry, my project was abandoned.
I was able to ADC sound, transfer it to remote esp via esp-now and DAC it again to the speaker. But quality was below expectation. Maybe because my faults and misuanderstanding. Source code no more exist :(