earlephilhower/ESP8266Audio #346: NoDAC PWM output stage

msalat, 2020-12-10:
Instead of a single transistor capable only of 0,2 A continous current I would suggest an emitter follower (npn+pnp, eg. bc337+bc327 (0,8 A)) and a capacitor (470 µF) to remove dc component of the signal.

Here the result of a spice simulation: red = pwm out, yellow = your "1 T" circuit, blue = emitter follower:

[image]

msalat, 2020-12-10:
sorry, current is not the problem here: with +- 0.7 V / 4 Ohm = 0.175 A

jgrulich, 2021-01-26:
This approach is the only correct one. The coupling capacitor is mandatory at all cases. The NoDAC output has DC offset of ~1.5V and when used any kind of single transistor, or DC coupled signal path to speaker, there is permanent DC current 1.5V/4Ohm=0.375A which cause heating of the speaker coil and may even cause permanent damage of it.

msalat, 2021-01-31:
Funnily the speaker used on the "bbc micro:bit v2" uses a circuit very similar to the 1 transistor circuit: https://datasheet.lcsc.com/szlcsc/1811151451_Jiangsu-Huaneng-Elec-MLT-8530_C94599.pdf

jgrulich, 2021-02-01:
Hi @msalat , this is NOT the same circuitry. In case of NoDAC there is square signal of ~60kHz and audio signal is done by changing of pulse width which changes the average voltage, which means PWM = Pulse Width Modulation.
In case, you've provided, the 5V square signal is directly with frequency of audible signal of 2.7kHz. This means 100% ON/OFF amplitude modulation, no PWM.