earlephilhower/ESP8266Audio #347: pcm5102 + esp32: I don't understand

fkoteam, 2020-12-13:
Hello,
my first atempt was using esp32 and internal DAC. There was a noise after playing my WAV's in PROGMEM. With the sample "viola" WAV's, all was OK. I tried fill my WAV's with " 0x00" at the end as "viola", but it wasn't work.
For the second atempt I want to use esp32 + PCM5102 but I have a doubt. Following #100, I think I have to connect:
BCK Gpio26
LCK Gpio25
DIN Gpio22
SCK GND

And use AudioOutputI2S. Is it correct? But it's the same class and pins I use with internal DAC. And the pins 25, 26 outputs "analog values". If I want to use PCM5102, I think that pins should output digital values. 
I don't understand.... Any help with my isue or this doubt?

Thank you!

fkoteam, 2020-12-13:
ok, I get it. To use an external DAC you have to change the contructor:

AudioOutputI2S(0,1);
to
AudioOutputI2S();

The connections to PCM5102 are:

    3.3V -> VCC, 33V, XMT
    GND -> GND, FLT, DMP, FMT, SCL
    BCLK (GPIO26 by default)->BCK
    I2SO (GPIO22 by default)->DIN 
    LRCLK (GPIO25 by default)->LCK 

Sound quality is better, but I still have a problem with some audio files. I attach two files. One with noise after playback and the other without noise. I think it's related with the size of file. Large files are ok, but short ones have noise at the end. Any idea?
[test_audio.zip](https://github.com/earlephilhower/ESP8266Audio/files/5685112/test_audio.zip)

troisiemetype, 2020-12-15:
The use of I2S is, as you said, quite straightforward : 
```
audioOutput = new AudioOutputI2S();
audioOutput->SetPinout(I2S_CLK, I2S_WS, I2S_DATA);
```
Maybe you could add a zip with the real files. My guess is that you have an ID3 section (or some other tag format) at the end of the file, and that this tag is read as sound. I have a similar problem with random buffer played at the beginning of mp3 files, while the file tag data is parsed.

fkoteam, 2020-12-15:
> The use of I2S is, as you said, quite straightforward :
>
> ```
> audioOutput = new AudioOutputI2S();
> audioOutput->SetPinout(I2S_CLK, I2S_WS, I2S_DATA);
> ```
>
> Maybe you could add a zip with the real [...]

Thank you very much. I think the files are plain WAV's. I applied a workaround adding silence at the end of the shortest files and it worked.

closed 2020-12-15 by fkoteam

troisiemetype, 2020-12-15:
> I think the files are plain WAV's.
They are wave data in a C table, that makes it difficult to read them as waves, open them with a tag editor, or interpret as hex files. ;)