earlephilhower/ESP8266Audio #348: Replace SD Card with SPI Flash

eugjr, 2020-12-13:
Hi, I'm studying a way to play wav files for a train model. Already got a prototype running with DFPLayer MP3 Board, but as the space in the train model is narrow, I need to miniaturize the assembly as much as possible.

Do you know if its possible to use your code, replacing the SDCard storage with a SPI flash like Winbond W25Q128? I don't have familiarity with SPI Flash chips, just saw some examples on WEB on how to store the files and acessing them via discrete adresses.
I found your code searching for the simultaneous playing options, because the assembly sometimes will need to run 2 sounds simmultaneous.

BugerDread, 2021-04-17:
I was also thinking about this and I found https://github.com/esp8266/Arduino/issues/2196 pointing to https://github.com/PaulStoffregen/SerialFlash but still did not tried to use with ESP8266audio

andy-preston, 2024-03-11:
You might find some useful information to help you here: https://hackaday.io/project/168177-micro-sd-card-to-spi-flash-adapter