earlephilhower/ESP8266Audio #351: where do we define sd pins for sd example?

manecolooper, 2020-12-20:
hi, i'm trying to test sd flac example with esp32, and don´t know where to define pins for sd card, i'm using a ttgo 1.7 that has 
CS 13  , MOSI 15 , SCK 14 and MISO 2 thanks!

blaudp79, 2021-02-20:
> hi, i'm trying to test sd flac example with esp32, and don´t know where to define pins for sd card, i'm using a ttgo 1.7 that has
> CS 13 , MOSI 15 , SCK 14 and MISO 2 thanks!

Hi, I have the same problem. Have you figured this out?

Kriegbaum, 2022-08-15:
I got mine working with the following: 
```
// SPI pins for SD card
#define SCK  7
#define MISO  6 
#define MOSI  5
#define CS  4

SPIClass spi = SPIClass(FSPI);

void setup(){
  spi.begin(SCK, MISO, MOSI, CS);
  pinMode(spi.pinSS(), OUTPUT);
  if(!SD.begin(CS, spi, SPI_SPEED)){
      Serial.println("Card mount failed!");
  }
}
```

I've omitted the rest of the garbage from my setup function

CobaltEcho, 2023-08-20:
For Reference, I dont think the pins are included in the callout (as a #define) because it assumes you are using the default SPI pins on the board.