earlephilhower/ESP8266Audio #353: M5STACK pinout

christiandelanus, 2020-12-23:
Hi.
I have some troubles changing pins from original values to m5stack core2. Have anyone do that?
Thanks a lot.

playground, 2021-02-20:
Will this work with Atom Echo Smart Speaker?

MisteryX, 2021-04-30:
It does not work on m5stack core2, because of missing support of JTAG on M5Stack platform and C/C++ API. Development on C/C++ for M5Stack is NOT recommended.

jarus0, 2021-07-10:
Hi there, 
First of all thanks for your awesome library.

I also have the same problem.
I'm using "M5Stack Core2" module and trying to play audio from an mp3 file but didn't work.
https://shop.m5stack.com/products/m5stack-core2-esp32-iot-development-kit

Any help would really appreciatable thanks.

jarus0, 2021-07-12:
Issue resolved, Simple pin change needed!
Thanks