earlephilhower/ESP8266Audio #355: Stuttering on ICY stations.

cjwhoishe, 2020-12-28:
Hello there,
I would like to thank you very much for this library and the work you have put into it.

I am trying to use an ESP32 board with its internal DAC to create an 'internet radio' solution, which I can fit into an old radio (AM only) to give that radio a new life. 
This library seemed to fit the bill by using the AudioFileSourceICYStream , AudioFileSourceBuffer, AudioGeneratorMP3 and AudioOutputI2S classes.
When I built it I found that some stations stuttered and even failed to 'sync'.
The stuttering was bad when first connecting and when some served web pages were being called.
(first connection stuttering was highlighted in an earlier issue #197   )
 I tried various thinks like:-
1. changing buffer sizes.
2. creating an individual freertos tasks to run the audio->loop in
3. giving this task a higher priority theoretically above the web page serving.

In the end it came down to 2 issues:
1. The DMA buffer used to transfer the I2S data was emptying
2. The code had real difficulty 'synching' to the ICY data blocks

The DMA buffers emptying was occuring when a 'largish' web page was being served and the effect was reduced by making use of the dma_buf_count in the creation of the AudioOutputI2S class being used.
ie I used 
new AudioOutputI2S(0, 1, 128);	// this is the maximum that is allowed
and not
new AudioOutputI2S(0, 1);	// this sets dma_buf_count to 8
I then had to be carefull about the size and length of time in serving web pages.

The difficulty to 'synch' was more interesting and the previously raised issue gave a clue.
The problem was due to the fact that the code in AudioFileSourceICYStream::readInternal() can only handle 1 complete ICY block wrapped with mp3 data or 1 block split over 2 calls.
If you have a fast internet connection that can ensure there is enough data available to contain a complete ICY block and part of a 2nd ( or even more) then the syncronisation goes all wrong.
I fixed this by restricting the amount of data processed by the call to less than an ICY block size by inserting the line
  if (len > ((icyMetaInt >> 1) )) len = ((icyMetaInt >> 1));
at the start of the function directly before the retry: label
This meant that all the buffer space could be used and it synchronisd more reliably.

Can you consider at least adding this to the library as it is a lot easier than trying to re-write the ICY decoder.

I am also look into re-writing the ICY decoder to handle this scenario, which is what you thought needed to be considered in the previous issue.

earlephilhower, 2020-12-28:
Thanks for the debug!  I've written it slightly differently, do you think you could try the change in #356 and report back?

cjwhoishe, 2020-12-29:
Hi,
Thanks very much and for confirming that the 'feature' was as I thought!
I have tried this both with and without using AudioFileSourceBuffer and it is fine.
I also tried it with buffer sizes of 768 (definitely less than 1 ICY block), 2048, 8192, 32768 and even 262144 (using PSRAM) when using AudioFileSourceBuffer.

The stream I was having trouble with was:
http://tx.planetradio.co.uk/icecast.php?i=jazz.mp3
This is a front end to a page which handles stream balancing and returns a 'found' redirect which can vary but an example is
http://edge-baueral-01-cr.sharp-stream.com/jazz.mp3 
I have handled these redirects outside of the AudioFileSourceICYStream class.

This never synchronised on my setup and has a metasize = 1024 and bitrate = 112Kb
It is one of many commercial radio stations within the UK with this type of characteristic stream.

I have found that my internet connection has the following
download speed - max 95Mb min 64Mb typical 80Mb
Latency - max 45ms min 10us typical 26ms
Jitter - worst 9ms

With my connection I probably do not need a buffer bigger than 1024 which is equivilant to 73ms @ 128Kb
This does in fact work OK, but I have settled on 8192 which is equivalent to about 1/2 second.

This has a drawback that the ICY data comes out 1/2 second early, consequently the display shows that the track has ended before it does.
This is obviously made worse when a bigger buffer is used and if I use 262144 (PSRAM) it is about 18 Secs early.

I'm wondering whether to buffer everything and do the ICY decode on reading from the buffer when considering the ICY decode mods.
Although as it is now working I may well leave it alone and get on with actually fitting it into a radio.

earlephilhower, 2020-12-29:
Thanks for the report back!  I do 99% of my work on the 8266, so never really considered the case where a buffer could be so large.

Thinking out loud, for the too-early reports, can you not just change the order of the filters:  HTTPsource->Buffer->ICY->MP3Decode?

commit 79515db7 (2020-12-28): Ensure only 1 ICY packet processed per reading

Thanks to @cjwhoishe, fixes #355 .

When a large enough read size was requested it is possible that for some
streams multiple ICY blocks could appear.  The logic doesn't support
that, so only read up to 1/2 the ICY block interval to ensure there is
no way to see two ICY blocks in one readInternal call.
PR #356 (2020-12-29): Ensure only 1 ICY packet processed per reading
Thanks to @cjwhoishe, fixes #355 .

When a large enough read size was requested it is possible that for some
streams multiple ICY blocks could appear.  The logic doesn't support
that, so only read up to 1/2 the ICY block interval to ensure there is
no way to see two ICY blocks in one readInternal call.
commit 7cb02a69 (2020-12-29): Ensure only 1 ICY packet processed per reading (#356)

Thanks to @cjwhoishe, fixes #355 .

When a large enough read size was requested it is possible that for some
streams multiple ICY blocks could appear.  The logic doesn't support
that, so only read up to 1/2 the ICY block interval to ensure there is
no way to see two ICY blocks in one readInternal call.
closed 2020-12-29 by earlephilhower