earlephilhower/ESP8266Audio #362: HTTPC_FORCE_FOLLOW_REDIRECTS not declared

DanDor90, 2021-01-17:
Board: Wemos D1 mini (ESP8266)

When running any http-related examples, I get the following error:

\Arduino\libraries\ESP8266Audio-master\src\AudioFileSourceHTTPStream.cpp:43:27: error: 'HTTPC_FORCE_FOLLOW_REDIRECTS' was not declared in this scope

   http.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);

Am I missing any particular library?

d-a-v, 2021-01-17:
Did you try to update your esp8266/arduino core to the 2.7.4 version ?

DanDor90, 2021-01-18:
Thanks a lot, that fixed the problem. I was running an older version of the esp8266 core on the Arduino IDE. Upgrading to 2.4.7. solved the problem and I was able to compile.

closed 2021-01-18 by DanDor90

xitonghan, 2024-03-27:
updata is ok!
thx