earlephilhower/ESP8266Audio #364: AudioOutputI2SNoDAC() no PWM at output with no data

jgrulich, 2021-01-26:
I'm using digital PWM Class-D amplifier directly connected at the RX pin with AudioOutputI2SNoDAC which works excellently, but facing one big issue.
When created object out = new AudioOutputI2SNoDAC(); the I2S module is initiated, blue LED at board is ON, but no PWM output at RX pin. It appears only when mp3->begin(buff, out); with valid data. When mp3 stops data generation, the PWM at RX pin disappears and there is 0V. This PWM on/off switching cause DC offset dropping from 0V to 1.5V and result is current burst at the Class-D amplifier output stage and terrible audible click at the speakers. In my case it is click with power of 300 Wats !!! I'm using mute signal for Class-D amplifier which is enabled with delay after mp3->begin(buff, out); which works well without any click, but there is no info before mp3 stops do have enough time to mute it again before PWM disappears.
Has anybody any idea how to correct the AudioOutputI2SNoDAC lib to keep the PWM running all the time?
I was using external DAC with analog output, there was no click, but due to analog way there was significant noise in the signal.
100% digital signal path is the only solution.

Jarvis785, 2021-01-30:
Hi @jgrulich 

even I'm facing similar issue when I'm playing same mp3 repeatedly in loop I keep hearing a click/bullet noise when the audio is starting/ending.

can you help me how to remove the click sound ?

void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop())
    {
      mp3->stop();
      delete mp3; delete id3; delete file; delete out; // free up memory !
      file = new AudioFileSourceSPIFFS("/pno-cs.mp3");
      id3 = new AudioFileSourceID3(file);
      id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
      out = new AudioOutputI2SNoDAC();
      mp3 = new AudioGeneratorMP3();
      mp3->begin(id3, out);
    }
  }
}

jgrulich, 2021-02-03:
@Jarvis785 This is the same issue. It's question of the AudioOutputI2SNoDAC lib. You may create the output only once at the setup{}, but the result is the same. There is missing PWM output when mp3 object doesn't exist.

Jarvis785, 2021-02-03:
@jgrulich is there any way to fix this ?

jgrulich, 2021-02-04:
@Jarvis785 Yes, fix it in AudioOutputI2SNoDAC lib.

Jarvis785, 2021-02-04:
@jgrulich that means you tried to fix but it's not fixed completely or what ? Is there any workaround ?

If you have the fixed code can you please share ?

jgrulich, 2021-02-04:
@Jarvis785 yes, I've tried to modify AudioOutputI2SNoDAC.cpp, but no success. I don't have any idea how it uses the internal I2S system to generate the PWM. Probably some part of the ESP core which is not well documented.

jgrulich, 2021-02-04:
@Jarvis785 Guess, that it will be in this part:
`bool AudioOutputI2SNoDAC::ConsumeSample(int16_t sample[2])`
There is mentioned this: "// Either send complete pulse stream or nothing" and when there are no stream data, it "send nothing".

earlephilhower, 2021-02-04:
I think there's a basic misunderstanding of how the library works.  This isn't a freerunning task or thread like on a PC, it's pumped by calls to mp3->loop() (in this case).  And, in the NoDAC case, we're doing a lot of math to generate a delta-sigma bitstream from standard signed ints coming in the pipe, too.

DAC or no DAC, the I2S hardware sends out data DMA'd to it.  When no new MP3 data is generated, it sends 0s (underflow).  When I2S->stop() is called at the end of an MP3, even that is stopped and you're back to a standard GPIO pin.  That would give you a pop/click on end/restart.

There's no general way of having it send something else w/o, say, hacking the mP3 decoder to never end and just send some neutral samples down the I2S pipe when completed.  You could hack in a flag saying when the MP3 was really done decoding and examine that in your main loop to see if it's safe to switch to another one.

closed 2021-02-04 by earlephilhower

jgrulich, 2021-02-04:
@earlephilhower Thank's for deep explanation. Now it's more easy to understand how it works. You've mentioned, that when MP3 stops, the RX pin is standard GPIO, but it is still output, not input as it need to be. When it will be input, there may be used simple 2R divider from Vcc and Gnd to keep there average DC level and the pops/clicks will finally disappear for ever.

Jarvis785, 2021-02-04:
Hi @jgrulich did the click soumd issue got resolved for you ? If yes please share the solution what's the 2R driver ?

jgrulich, 2021-02-04:
Try this way: Just after `I2S->stop()` put RX as input and put 10k resistor from Vcc to RX pin and another 10k resistor from RX pin to Gnd. This will work for sure only for signal output with low-path filter and amplifier, or for 2T and cap. output, not for basic 1T output.

Jarvis785, 2021-02-04:
@jgrulich Thanks for the advice!
One more question.. whats 2T again ?

jgrulich, 2021-02-04:
It is PNP-NPN emitter follower and cap. 470uF in series with speaker. 1k resistor from RX pin to 2T base may be used too for protection reasons, but is not mandatory. As of 2T may be used some BC 3xx or similar 500-800mA signal BJTs.

Jarvis785, 2021-02-04:
@jgrulich if you have any diagram you can attach then it will be better. Otherwise i will try to implement it as oer my understanding

jgrulich, 2021-02-04:
sorry, don't have schematic at hands.

jgrulich, 2021-02-05:
@Jarvis785 good news, I've tested the 2R solution with RC low path filter and it works well, no pops/clicks any more !!!
@earlephilhower please be so kind and put me the right direction how to make the RX pin the fastest switching between I2S output and GPIO input, or regular RX pin when no I2S data. The 2R divider works, but the pin need to be input to keep the DC voltage there.
Now I'm using
`
if (!mp3->loop())
{
mp3->stop();
delete out;
pinMode(RX, INPUT);
}
 `
but it's too slow and there is long 0V at RX pin before it becomes input and 2R divider restores the DC offset.

Jarvis785, 2021-02-05:
@jgrulich only 2R setup you explained earlier is enough ? Or RC low path filter is also required ?
I'm a newbie here, if you could help me with a rough diagram with exact resistor and capasitor values tgen it will be great !
Thank you in advance !

earlephilhower, 2021-02-05:
> @earlephilhower please be so kind and put me the right direction how to make the RX pin the fastest switching between I2S output and GPIO input, or regular RX pin when no I2S data. The 2R divider works, but the pin need to be input to keep the DC voltage there.

In the AudioOutputI2SNoDAC, in the `stop()` method just add the pinmode right after i2s_stop().

jgrulich, 2021-02-05:
Here is the schematic of the LP filter with the DC restore. The output may be used for any audio amplifier, or active headphones.

[image]

Jarvis785, 2021-02-05:
@jgrulich you are a true lifesaver !

jgrulich, 2021-02-05:
@earlephilhower sorr, I've not found i2s_stop(), or stop() method in the AudioOutputI2SNoDAC.
Do you mean this `i2s_end()` in AudioOutputI2S.cpp?
`
AudioOutputI2S::~AudioOutputI2S()
{
  #ifdef ESP32
    if (i2sOn) {
      audioLogger->printf("UNINSTALL I2S\n");
      i2s_driver_uninstall((i2s_port_t)portNo); //stop & destroy i2s driver
    }
  #else
    if (i2sOn)
      i2s_end();
  #endif
  i2sOn = false;
}
`

jgrulich, 2021-02-05:
seems, that `i2s_end()` is somehow not working, because otherwise it will change to input mode itself.
In core_esp8266_i2s.cpp is this i2s_end() function:
`
  if (tx) {
    pinMode(I2SO_DATA, INPUT);
    pinMode(I2SO_BCK, INPUT);
    pinMode(I2SO_WS, INPUT);
    free(tx);
    tx = NULL;
  }
`

earlephilhower, 2021-02-05:
That's the destructor, so unless you `delete i2s` it's never going to be called.  If you don't delete the i2s, then the I2S is still clocking bits out since it underflows.

jgrulich, 2021-02-05:
Yes, I've found the same. The pin is released only by `delete out;` even there are no data. May be the easier way will be to detect underflow and when it occurs, than perform `delete out;` But I don't know how it will act when the buffer is filled and there will be missing out module.

jgrulich, 2021-02-05:
Confirmed, I've removed the pinMode and used only this:
`if (!mp3->loop()) { mp3->stop(); delete out; }`
and it works the same way. `delete out` really destroy the whole i2s module and returns pins to input mode.
But there is still some delay from the last i2s clock and `delete out` is performed.
How to detect that i2s will underflow? It will be useful not only for this case, but even for regular i2s DAC to control the amplifier MUTE function, output protections and so on.

earlephilhower, 2021-02-05:
For normal I2S DACs, 0 is a nice safe silence number, but when we're hacking out delta-sigma it's actually -INFINITY.  So for normal use the core is doing fine, but in this case it's definitely not optimal.

You could see if the I2S HW has an underflow interrupt (I don't think so, but you'll have to dig thru what docs you can find to confirm.)  Might also look at the DMA engine and see if it has an IRQ on end of chain which would could also be used (since if DMA hits the end, there's underflow for sure).

jgrulich, 2021-02-05:
@Jarvis785 try this modified code and the schematic above, guess it will work without pops and clicks.
`
void loop() {
if (mp3->isRunning()) {
if (!mp3->loop()) {
mp3->stop(); delete mp3; delete out; delete id3; delete file; // discard all old objects
file = new AudioFileSourceSPIFFS("/pno-cs.mp3"); // create the objects again
id3 = new AudioFileSourceID3(file);
id3->RegisterMetadataCB(MDCallback, (void*)"ID3TAG");
out = new AudioOutputI2SNoDAC();
mp3 = new AudioGeneratorMP3();
mp3->begin(id3, out);
}
}
}
`

Jarvis785, 2021-02-05:
> @Jarvis785 try this modified code and the schematic above, guess it will work without pops and clicks.
> `void loop() { if (mp3->isRunning()) { if (!mp3->loop()) { mp3->stop(); delete mp3; delete out; [...]

Hi @jgrulich when I'm comparing this code with the code I've shared with you earlier, i don't see any difference, just this time i'm gonna use the schematic. Am I missing something here ?

jgrulich, 2021-02-05:
Yes, there is different order of objects delete functions. `delete out` is put just after mp3 is destroyed to release the pint to input state as fast as possible. `delete id3; delete file` is not time sensitive and may wait.

Jarvis785, 2021-02-06:
Hi @jgrulich I made this circuit as per the schematics, still the clicks/pops is not fixed. 
I have added the audio recording, you can hear the click sound when the mp3 ends/starts. I'm using esp8266-12E, which hardware are you using ? How it got fixed for you !! ?

[Audio Record 1.zip](https://github.com/earlephilhower/ESP8266Audio/files/5936368/Audio.Record.1.zip)

[image]

jgrulich, 2021-02-06:
Than you need to use scope to see the signal timing. What are you using as a signal amplifier? What's its input impedance?
I'm using ESP01S and XH-M125 mini amplifier board. Instead of R4 at the schematic above I have 50k pot. for volume control.
Sometimes there is small pop at the end of the stream, but nothing compared to the original state. I've looked at the RXD data and this is cause by the software. There is probably some buffer emptying which cause that data stop for 200-500ms, than restart for 24ms and than finally stop. And this 24ms burst cause the pop. If the stop is clear, no pop is there.

Jarvis785, 2021-02-06:
> Than you need to use scope to see the signal timing. What are you using as a signal amplifier? What's its input impedance?
> I'm using ESP01S and XH-M125 mini amplifier board. Instead of R4 at the [...]

I'm using aux cable to connect rx and gnd to a Bluetooth speaker, also tried pc's external stereo speakers , still tje issue persists

jgrulich, 2021-02-06:
Look at the RXD data if there is the burst.

Jarvis785, 2021-02-06:
> Look at the RXD data if there is the burst.

Oops, I don't have a oscilloscope 
What do you expect from a newbie 🤣

jgrulich, 2021-02-06:
OK, supposed newbie with ESP-Audio. Than it will be more complicated, because there may be more sources of the pop-clicks and need to identify the right one. It may be irregular start/end of the file, buffer emptying, or may more. The file you can't check at PC, of in mobile, because there are high level players which have software filters and recover any file abnormality. But ESP-Audio is pure data player without any additional functionality. The file need to be 100% correct.

earlephilhower, 2021-02-06:
@jgrulich mind if I add your LPF to the docs?

jgrulich, 2021-02-07:
@earlephilhower not sure, need to be deeply tested. @Jarvis785 reported that it's not working.

jgrulich, 2021-02-07:
@Jarvis785 guess, that the best way for you is to use I2S output and MAX98357A board which is 3W Class-D audio amplifier with I2S input, means that you need only 3 wires to connect ESP and MAX, connect 4Ohm speaker to MAX and that's all. Without pops and clicks for ever.