earlephilhower/ESP8266Audio #368: Release of GPIO2 not working for 1.8.0 and 1.8.1 version

jgrulich, 2021-02-09:
I'm using ESP01S board and have occupied both GPIO0 and GPIO2 pins. Rev. 1.6 and 1.7 works well, but 1.8 doesn't release GPIO2 and keep there i2s clock.

msilenius, 2021-03-19:
I confirm the problem. Built in led turn on  while playing sound and stay on after  on nodecmu from version 1.8 up.

jgrulich, 2021-03-23:
No, this is not exactly correct. The GPIO2 allocation is done in AudioOutput class, not in AudioGenerator class.
This is the right approach:
`  out = new AudioOutputI2SNoDAC();         // Load audio output driver - it changes LED as I2S data output
  pinMode(LED_BUILTIN, OUTPUT);               // Restore the LED as GPIO output
`
Than until you keep the AudioOutput object, the output is keeped restored regardless on AudioGenerator object.
This is better in case when playing more files. AudioFileSource and AudioGenerator is discarded and created new, but AudioOutput is keeped unchanged.

msilenius, 2021-03-29:
actually it seems the same problem happen with GPIO 15

jgrulich, 2021-04-06:
Confirmed, 1.7 is working well.

closed 2021-04-06 by jgrulich